Ultimaker/Cura #12322: MacOS Catalina: File has been modified/Would you like to reload on every STL file import

solarbirdy, 2022-05-22:
### Application Version

5.0.0

### Platform

MacOS Catalina 10.15.7

### Printer

Ender 3 V2

### Reproduction steps

1. Drag any .STL file into Cura, when Cura is already open and running

### Actual results

After import, a dialogue appears titled "File has been modified" with contents "Would you like to reload [filename]?" with a reload button and a close-dialogue X in the upper right corner.

If you close the dialogue, everything works fine, but still.

### Expected results

No such dialogue box should appear.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

MacBook Pro Mid-2012, 2.9 GHz Dual-Core Intel Core i7, 16 GB 1600 MHz DDR3, Intel HD Graphics 4000 1536 MB.

solarbirdy, 2022-05-22:
[image]

devzu, 2022-05-24:
This also occurs in newer macOS versions(Monterey 12.2.1), both on M1 and M1 Pro chips.

Vandresc, 2022-06-07:
Hi @solarbirdy, @devzu , so no changes have been made in Cura at all that have modified the model that makes Cura trigger that message? Could you please share the logs with us and the file? Thank you!

solarbirdy, 2022-06-11:
"The file" _was_ literally any STL file I dragged into Cura 5. Literally any STL, I tried several across multiple sessions, and it reproduced every time, and it continued to repro after I reported the issue.

	But today I realised I hadn't noticed the dialogue for a while and I've also realised it's not just that I've adopted the muscle memory to dismiss it instantly - it's stopped reproducing and I have no idea why.

	Fortunately, my cura.log file goes back to when it was reproducing! So you'll find a big chunk of that from the relevant dates attached, along with an STL file that reproduced it before. I could pick basically any STL I have, but this is the original unmodified file from a project I was working on when I filed the report. It's mentioned several times in the log files, so hopefully it's a good example.

									- Dara

2022-05-21 22:06:14,702 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [97]: Adding QT5 plugin path: /Applications/Ultimaker-Cura.app/Contents/Resources/plugins
2022-05-21 22:06:14,702 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
2022-05-21 22:06:14,702 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 5.0.0
2022-05-21 22:06:14,703 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 8.0.0
2022-05-21 22:06:14,703 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type None
2022-05-21 22:06:14,703 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/lib/uranium must be a folder.
2022-05-21 22:06:14,703 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/lib64/uranium must be a folder.
2022-05-21 22:06:14,703 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/lib32/uranium must be a folder.
2022-05-21 22:06:14,704 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/MacOS/plugins must be a folder.
2022-05-21 22:06:14,704 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/Resources/uranium/plugins must be a folder.
2022-05-21 22:06:14,704 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/Resources/cura/plugins must be a folder.
2022-05-21 22:06:14,704 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
2022-05-21 22:06:14,704 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is /Users/kahvi/Library/Application Support/cura/5.0
2022-05-21 22:06:14,705 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is /Users/kahvi/Library/Application Support/cura/5.0
2022-05-21 22:06:14,705 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is /Users/kahvi/Library/Application Support/cura/5.0/cache
2022-05-21 22:06:14,705 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [502]: Found config: ['/Users/kahvi/Library/Application Support/cura', '/Users/kahvi/cura'] and data: ['/Users/kahvi/Library/Application Support/cura', '/Users/kahvi/cura']
2022-05-21 22:06:14,705 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [506]: Latest config path: /Users/kahvi/Library/Application Support/cura/4.13 and latest data path: /Users/kahvi/Library/Application Support/cura/4.13
2022-05-21 22:06:14,705 - INFO - [MainThread] UM.Resources.copyVersionFolder [542]: Copying directory from '/Users/kahvi/Library/Application Support/cura/4.13' to '/Users/kahvi/Library/Application Support/cura/5.0'
2022-05-21 22:06:14,705 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /Applications/Ultimaker-Cura.app/Contents/MacOS/share/uranium/resources/bundled_packages/uranium.json
2022-05-21 22:06:14,705 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/bundled_packages/cura.json
2022-05-21 22:06:14,706 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
2022-05-21 22:06:14,706 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
2022-05-21 22:06:14,706 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [178]: OpenGL renderer type for this OpenGL version: Intel HD Graphics 4000 OpenGL Engine
2022-05-21 22:06:14,706 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [180]: Detected most suitable OpenGL context version: 4.1 Core profile
2022-05-21 22:06:14,706 - INFO - [MainThread] UM.Qt.QtApplication.initialize [187]: Initializing job queue ...
2022-05-21 22:06:14,706 - INFO - [MainThread] UM.Qt.QtApplication.initialize [191]: Initializing version upgrade manager ...
2022-05-21 22:06:14,706 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /Applications/Ultimaker-Cura.app/Contents/MacOS/share/uranium/resources/bundled_packages/uranium.json
[... cut ...]
2022-05-21 22:07:22,510 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [494]: Loaded plugin SolidView
2022-05-21 22:07:22,526 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [494]: Loaded plugin FirmwareUpdateChecker
2022-05-21 22:07:22,536 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [494]: Loaded plugin SupportEraser
2022-05-21 22:07:22,543 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [494]: Loaded plugin PrepareStage
2022-05-21 22:07:22,558 - INFO - [MainThread] UM.PluginRegistry.l

solarbirdy, 2022-06-13:
It just reproduced again. This was in a longer session where I'd generated several .gcode files with small variations from the same .stl file over time for a printing test. Then I loaded a different stl and the problem recurred.

	Log file and .relevant .stl file attached.

solarbirdy, 2022-06-16:
hey, we're back to reproducing pretty freely, here's another repro for you with a different stl file

[image]
[Wheel.stl.zip](https://github.com/Ultimaker/Cura/files/8923315/Wheel.stl.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8923317/cura.log)

solarbirdy, 2022-06-16:
This repro is interesting and slightly different because it did NOT show up upon loading the .STL file, but instead showed up only after slicing the object:

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8923374/cura.log)
[Mount_B.stl.zip](https://github.com/Ultimaker/Cura/files/8923376/Mount_B.stl.zip)
.

epsalmond, 2022-09-21:
Confirming that https://github.com/Ultimaker/Uranium/pull/852 fixes this.

closed 2022-09-30 by Joeydelarago

borthick, 2023-02-23:
This bug is still occurring. I've found that disconnecting from internet before opening Cura prevents it from happening. Why would having an internet connection cause this?

mrbrdo, 2023-12-19:
Having this problem as well for a long time now.

Joeydelarago, 2023-12-19:
Hey @mrbrdo, I stopped working an UltiMaker earlier this year. If you still see this issue in the latest version, then please open Cura immediately do the action that causes this, close cura, and upload the logs here (This makes it easier to find the logs around your issue).

Perhaps @MariMakes can bring the issue to the team for you.

solarbirdy, 2023-12-23:
Coming back to this after ages: I see this behaviour very regularly in MacOS on M1 (Apple MacBook Pro M1 Pro 16GB running several versions of Sonoma but currently 14.2)

BUT

I can't reproduce it right now in a single load/slice/preview path as it usually _seems_ to do. Fiddling around with it before a slice isn't making it happen at the moment either and so on.

I _suspect_ that it's something that happens more often as you leave Cura open longer because I don't close it between print trials; I just keep it loaded while I see how the print goes and then modify later once I've got enough data to try again. So.

borthick, 2023-12-23:
Since I upgraded to the newest version, it now happens EVERY time I open Cura for the first time, whether I have internet or not. I immediately close and reopen, and then get no error. It's not really a matter of doing anything to reproduce it, since it happens every time. (Macbook Air 2017)

solarbirdy, 2023-12-23:
Got it. Happened first time I tried to use Cura this morning on a very simple STL object.

[cura.log](https://github.com/Ultimaker/Cura/files/13760362/cura.log)