Ultimaker/Cura #12324: [5.0.0] Gap between infill and wall when connecting 2 infill polygons.

KryukovDS, 2022-05-23:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Drag the model to the slicer.
2. Slice the model.

### Actual results

A gap appears between the filling and the inner wall.

### Expected results

There was no gap in Cura v4.13.0.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Project_&_Log.zip](https://github.com/Ultimaker/Cura/files/8752209/Project_._Log.zip)

[image]

[image]

Vandresc, 2022-06-07:
Hi @KryukovDS , thank you for your feedback. I will bring this up with the rest of the team.

nallath, 2022-06-10:
As we don't think that this is too big of an issue and we have a lot of other issues on our backlog, we've decided to defer this issue.

kvnper, 2022-07-09:
Also having this issue

[image]