Ultimaker/Cura #12325: Octoprint error message not closable

kylane, 2022-05-23:
### Application Version

5.0.0-beta1

### Platform

Win11

### Printer

Ender 3

### Reproduction steps

Opening Cura without turning on my octoprint first shows an error on load that I cant close until I start octoprint.

[image]

### Actual results

Cura error opens offscreen making it impossible to dismiss.

### Expected results

Popover panel should still be closable - perhaps scroll the contents within?

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

NO need - easily reproducable.

fieldOfView, 2022-05-23:
Issues with the OctoPrint Connection plugin should go here:
https://github.com/fieldofview/Cura-OctoPrintPlugin/issues

One could argue that Cura should handle messages that are too long in a better way.

fieldOfView, 2022-05-23:
Note that you are using a beta version of Cura 5, and the full release has been out for a while now. Please update.

nallath, 2022-05-23:
> One could argue that Cura should handle messages that are too long in a better way.

I figured that was also what this bug was about.

MariMakes, 2023-05-15:
Quick update, 
We discussed it with the team and see value in making long messages scrollable so they can be closed, but we totally forgot to update this issue. 

For internal reference CURA-9374

Thanks for the report! 👍

MariMakes, 2023-08-25:
Hey @kylane,

Quick update from our side 👋

Our new developer @wawanbreton was able to resolve your issue here: https://github.com/Ultimaker/Cura/pull/16509
Long messages now have a scroll bar. [image]

I'll be closing this issue and you will see the fix in the upcoming release.

closed 2023-08-25 by MariMakes

kylane, 2023-10-17:
> Hey @kylane,
> 
> Quick update from our side 👋
> 
> Our new developer @wawanbreton was able to resolve your issue here: #16509 Long messages now have a scroll bar. [image]
> 
> I'll be closing this issue and you will see the fix in the upcoming release.

Awesome work!