Ultimaker/Cura #12327: Marketplace button not working

vegeele, 2022-05-23:
### Application Version

5.0

### Platform

Mac Monterey

### Printer

UM S5

### Reproduction steps

1. Open Cura 5
2. Sign in to my account
3. Open and load a model
4. Attempting to change materials in Cura to something different than what is loaded on my S5 by clicking on the loaded materials
5. Get the error message - extruder 2 is disabled, no material loaded
6. Marketplace button doesn't work anymore
7. The same steps without triggering the error message; marketplace button works.

### Actual results

You click the Marketplac button, marketplace doesn't load

### Expected results

Marketplace should load

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

nallath, 2022-05-23:
I can't reproduce this issue. Could you include the log files, maybe they contain some extra information that shows what is going wrong.

vegeele, 2022-05-24:
Today I also couldn't reproduce it at first, but later it happened again. I then noticed the marketplace tab was hidden behind the Cura window. What I suspect what happened yesterday is that it might have loaded with a delay, and after clicking the marketplace icon twice, it basically hid/moved the marketplace window behind the Cura window and that is why it didn't respond / bring it to the front. Perhaps a UX improvement we can do here..

Vandresc, 2022-06-09:
Hi @vegeele, so the issue is that the marketplace window is placed behind Cura when double-clicking the icon? Is that correct?  If that is the case, we did have this discussion with our UX'er in the past and it is currently not on our backlog due to other high prio tickets.  @lorenzoromagnoli any plans for the future? I will defer this ticket if the issue is the hidden window when double-clicking, but if I misunderstood please re-open. Thank you!

closed 2022-06-09 by Vandresc