Ultimaker/Cura #12330: Support blocker turns to solid

Weeperr, 2022-05-23:
### Application Version

5.0.0

### Platform

Mac OS 11.6 Big Sur

### Printer

Creality CR-10s

### Reproduction steps

I click on support blocker and then click the highlighted object and the small support blocker square shows up, I select "settings" under the "mesh type: normal model" option and select Infill, I change the infill to 23% and resize the cube to envelope the part where I'd like the infill to vary from the master settings and select slice.

### Actual results

The support blocker turns into a solid object which would also be printed along with the object 

### Expected results

In the beta of 5.0.0 and the previous full version I was able to do the same steps and have one print with two different infill densities. Instead of the support blocker becoming a solid object , In this case a giant rectangular prism that isn't a part of the design but is meant to section off the design so that individual settings can be applied to that section of the designt. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I'm not allowed to share these designs. Hopefully this information shared is enough to sort out the issue.

smartavionics, 2022-05-23:
>  I select "settings" under the "mesh type: normal model" option

You need to use Modify Settings for Overlaps, not normal model.

Here's a little example where I put a support blocker into the side of another model and frigged its infill pattern and density...

[image]

Weeperr, 2022-05-24:
thanks for this! It worked for the most part but I'm curious as to why I was able to do it the other way previously. The aspect in which it didn't work was using the support blocker to create "fuzzy skin" while also using it to have an alternate infill setting. The infill worked but the fuzzy skin didn't.

closed 2022-06-10 by nallath