Ultimaker/Cura #12331: Feature request: Adaptive layer height or volumetric fan modifier

tastyratz, 2022-05-23:
### Is your feature request related to a problem?

When printing objects in petg at 0.3 layer height, I can typically do so well with 15% fan. When adaptive layers allows for much thicker layers I find cooling insufficient. At .48 LH I might need 20-22% fan. If I just set 22% fan to begin with, I'd reduce layer adhesion. This will compound on detailed sections of the part with thinner layers. Constant fan speed could cause uneven cooling over the print when considering different volume/btu per layer.

### Describe the solution you'd like

Even cooling for even volumetric flow or layer height consideration seems best. Bridge settings and other fan modifiers could overrule it, but, the ability to set a modifier bound to adaptive layer height seems easiest.

I.E. I envision a setting
Adaptive layer height fan multiplier: 30% (default) 
This setting will modify fan speeds up to 30% based on layer height variations deviating from the base layer height. I.E. a fan speed of 10% and adaptive layer modifier will change to 13% or 7% depending on Cura layer height deviation.

### Describe alternatives you've considered

Manually adjusting fan speeds from the printer control panel after I start seeing problems.

### Affected users and/or printers

Everyone

nallath, 2022-06-10:
I will discuss the feature request with the rest of the team!

nallath, 2022-06-21:
Unfortunately we've decided to defer this feature request for now. We do believe that there is value in this feature request, but we simply have too many improvements to choose from. As such we have to make tough choices like this.

If someone would like to have a go at this themselves, we would gladly welcome a pull request.

tastyratz, 2022-06-23:
I think I forgot to mention adaptive line width in my original post as it appears now in cura 5. This significantly compounds the issue combined with layer height and makes for a massive volumetric flow difference.

Maybe a multiplier with average line width for a layer versus configured line width and layer height vs configured layer height could be a simpler starting change? Not sure.

I believe it will become a greater issue over time now that we're seeing how much line width can be pushed on e3d style nozzles with wide flats.  200% wider printed at double height is a huge volumetric change for BTU put into the part.