Ultimaker/Cura #12332: Multiple different printers at the same time

johnReeferVermont, 2022-05-23:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Creality 5 and 6

### Reproduction steps

When I flip between Ender 5 and Ender 6 and then slice, what is sliced prints horrible.  It behaves the same way on 5.0.  It seems to work ok when I change to the printer I want, close and restart Cura.  

### Actual results

When you change from one printer to another then slice again, what is sliced prints horrible.  Same on Cura 5.0

### Expected results

I should be able to change printers without having to deal with a bad slide.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing more to report

GregValiant, 2022-05-25:
Did you check the settings profile, the material, and nozzle sizes after switching?  If you had the slice all set up for the E6 when you changed printers then the material/nozzle/profile might have changed to whatever the last were for the E5 and over-written your setup.

johnReeferVermont, 2022-05-25:
Actually I figured out what the problem was.  It had nothing to do with profiles.  My problem is that when check Print Tiny Walls, the print comes out awful.  It is a desing with posts and the posts are about 1 mm.

Sent from Note 20 Ultra

closed 2022-06-15 by github-actions[bot]