Ultimaker/Cura #12333: Printer Start G-Code Will Not Save.

ciley417, 2022-05-23:
### Application Version

Cura 5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3 V2

### Reproduction steps

I added a line of G-Code to Start G-Code under Printer Settings/Printer.  
After making the change and closed the code I added didn't save.

I also tried to copy the G-Code to the clipboard to use elsewhere.

I also noticed there is no **Close** button located at the Bottom right like in Cura 4.13.1.

### Actual results

The G-Code I added didn't save like it does in Cura 4.13.1.  I also noticed there is no **Close** button located at the Bottom right like in Cura 4.13.1.

I was unable to copy the G-Code to the clipboard.

### Expected results

I expected it to save my changes.

I expected it to copy to the clipboard like in Cura 4.13.1.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have no additional information.

GregValiant, 2022-05-23:
This is clumsy but right now it is what it is.
There are no plans for a close button and there is no right click menu in either the StartUp or Ending Gcode boxes.
Use CTRL+C to copy and CTRL+V to paste.  You will be unable to tell what text you have selected because it doesn't get highlighted.
After making changes to the StartUp Gcode - click in the Ending Gcode box before you close the dialog.  That will make the change.  (I think that was supposed to have been fixed for the 5.0.0 release??)
This is all do to the change from Qt5 to Qt6 that controls the dialog boxes.

Sonicboom247, 2022-05-25:
still an issue, it actually dumped my changes after a few prints, i was wondering why all of a sudden my printer didn't tilt the mesh when starting a print, i go check and my 
M117 Homing and Leveling
G29 L0 ;Load Mesh slot 0
G29 A ;Activate UBL
G29 J3 ;Probe Bed for mesh tilting
are gone.

Vandresc, 2022-06-07:
Hi @ciley417 , this is already fixed for the next release. Will close the issue but if the issue still persists in the new release as well please feel free to re-open the ticket. Thank you all for your feedback!

closed 2022-06-07 by Vandresc