Ultimaker/Cura #12338: Stuttering in models with many curves

luizbgomide, 2022-05-24:
### Application Version

5

### Platform

Windows 10

### Printer

Two Trees Bluer

### Reproduction steps

1. Sliced the file https://www.thingiverse.com/thing:4162772 with default slicing settings (except for initial bed layer .3 and 4 walls to make the print solid).
2. Printed it through Octoprint.
3. I used ArcWelder (v1.1.0rc4 ) to try to make make it smaller (around 50% of reduction in size).
4. Printed it again.

### Actual results

I got a lot of micropauses during the print (to the point that the expected time from PrintTimeGenius was 1h14m and the actuall time was 1h48m), in both slices (with and without ArcWelder, with AW the print time was bit shorter 1h34m) which produced lots of blobs on the surface.

### Expected results

No micropauses and no blobs.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Result print:
[image]

Produced Gcode: [Cable Guide.zip](https://github.com/Ultimaker/Cura/files/8762339/Cable.Guide.zip)

Produced Gcode with AW: [Cable Guide.aw.gcode.zip](https://github.com/Ultimaker/Cura/files/8762343/Cable.Guide.aw.gcode.zip)

luizbgomide, 2022-05-24:
@GregValiant the default settings (0.25mm):
[image]

GregValiant, 2022-05-24:
I looked at the The TwoTrees base definition file and it lists the default at:
"meshfix_maximum_resolution": { "value": "0.25" }
That may be too small a number (Creality printers had that problem with their definitions).  Try setting it to 0.5 and re-slice and try to print one of those again.

luizbgomide, 2022-05-24:
@GregValiant I changed the Maximum Resolution to 0.5 and recreated the file. The file size indeed got shorter (about 33% of the original size, both straight out of Cura and after AW).

Those are the original reported segments (**resolution 0.25**):
```
Extrusion/Retraction Detail
┌─────────────────────────────────────────────┐
│   Min          Max     Source  Target Change│
├─────────────────────────────────────────────┤
│  0.000mm to   0.002mm       0       0   0.0%│
│  0.002mm to   0.005mm       1       0-100.0%│
│  0.005mm to   0.010mm      13       8 -38.5%│
│  0.010mm to   0.050mm     187      75 -59.9%│
│  0.050mm to   0.100mm     537     354 -34.1%│
│  0.100mm to   0.500mm  114269   53346 -53.3%│
│  0.500mm to   1.000mm   34740   12402 -64.3%│
│  1.000mm to   5.000mm     702    3639 418.4%│
│  5.000mm to  10.000mm     408    2601 537.5%│
│ 10.000mm to  20.000mm     268     708 164.2%│
│ 20.000mm to  50.000mm     638     960  50.5%│
│ 50.000mm to 100.000mm       0       0   0.0%│
│          >= 100.000mm       2       2   0.0%│
├─────────────────────────────────────────────┤
│Total distance source:............87391.783mm│
│Total distance target:............87408.683mm│
│   Total count source:.................151765│
│   Total count target:..................74095│
│ Total percent change:.................-51.2%│
└─────────────────────────────────────────────┘
```

Those are the new reported segments (**resolution 0.5**):
```
Extrusion/Retraction Detail
┌─────────────────────────────────────────────┐
│   Min          Max     Source  Target Change│
├─────────────────────────────────────────────┤
│  0.000mm to   0.002mm       0       0   0.0%│
│  0.002mm to   0.005mm       0       0   0.0%│
│  0.005mm to   0.010mm      11       4 -63.6%│
│  0.010mm to   0.050mm     162     136 -16.0%│
│  0.050mm to   0.100mm     237     120 -49.4%│
│  0.100mm to   0.500mm   12880    5276 -59.0%│
│  0.500mm to   1.000mm   87346   38221 -56.2%│
│  1.000mm to   5.000mm    1422    3583 152.0%│
│  5.000mm to  10.000mm     410    2600 534.1%│
│ 10.000mm to  20.000mm     269     652 142.4%│
│ 20.000mm to  50.000mm     635     925  45.7%│
│ 50.000mm to 100.000mm       0       0   0.0%│
│          >= 100.000mm       2       2   0.0%│
├─────────────────────────────────────────────┤
│Total distance source:............87331.607mm│
│Total distance target:............87346.028mm│
│   Total count source:.................103374│
│   Total count target:..................51519│
│ Total percent change:.................-50.2%│
└─────────────────────────────────────────────┘
```

Now, is there a way for me to check if Cura is actually adding pauses somewhere instead of just printing it and watching the result?

GregValiant, 2022-05-24:
The printer follows the gcode.  Octoprint can add certain commands (I think) but Cura just builds a gcode file for printer movement.
The normal pause commands are M0/M1 M600 (filament change) and M25 (pause an SD print) but they all require user intervention to resume (usually a button click).
G4 is a pause command (Marlin calls it a "dwell") and takes either a P parameter in milliseconds or an S parameter in seconds.  I'm sure Cura did not add any G4 commands and in fact there are none in the files.
I had started to write a book here.  Instead, I decided to print the "Cable Guide.gcode" file.  It's printed from an SD card and on my Ender 3 Pro.
I would suggest you do a print from the SD card just to take Octoprint out of the loop.  An alternative is that maybe your mainboard is just having trouble processing the data but it's more likely that something is going on between Octoprint and your printer.
Considering that the settings weren't optimized for my Ender 3 Pro I would rate the prints as acceptable.  They certainly don't show all the warts that appeared on your prints.  This doesn't look to me to be a Cura problem.
[image]
[image]

luizbgomide, 2022-05-25:
@GregValiant I changed the Maximum Resolution to 0.5 and printed it and got no blobs anywhere.

So I wouldn't say the issue is with Cure engine per se, but with its default profiles that have unrealistic levels of resolution.
This [post](https://www.cnckitchen.com/blog/removing-zits-and-blobs-on-your-3d-prints-adjust-your-slicing-resolution) from CNC Kitchen seems to indicate that this was also an issue with Creality profiles (which was probably the basis for the TwoTrees Bluer.

GregValiant, 2022-05-25:
The definitions are created by either the manufacturer of the printer or a member of the community.  Ultimaker does not create definitions for other peoples hardware.
The author of the Creality definitions was someone named "trouch.com".  The author of your printer definition is listed as "3DGadgets.my".  Looking just at the Ending Gcode it is word for word what is used for the Creality machines.  The Maximum Resolution in the Creality definitions was changed from .25 (totally unrealistic) to .4 in about Cura 4.12.

luizbgomide, 2022-05-25:
I will see if I can find where to change that to .4 in the code. If that works for Enders that should work for Bluers as well.

GregValiant, 2022-05-26:
It probably depends on the mainboard capabilities.
To change the value to 0.40 or 0.50 you would need to change this line in the "two_trees_base.def.json" file:
 "meshfix_maximum_resolution": { "value": "0.25" },

A downside to making a local change like that is the next time you upgrade Cura it will revert to the old value.  I keep my own definition files for my Ender.  When I upgrade Cura I copy my files in.  That way things like my customized Start and End gcodes are the "default".

luizbgomide, 2022-05-26:
@GregValiant I meant a fix to the Bluer profile in the Cura source code, the same way that was fixed in the Ender profile in 4,12.

GregValiant, 2022-05-26:
Most of the setting defaults are in fdmprinter.def.json.  Here is the default for Maximum Resolution:
 "meshfix_maximum_resolution":
                {
                    "label": "Maximum Resolution",
                    "description": "The minimum size of a line segment after slicing. If you increase this, the mesh will have a lower resolution. This may allow the printer to keep up with the speed it has to process g-code and will increase slice speed by removing details of the mesh that it can't process anyway.",
                    "type": "float",
                    "unit": "mm",
                    "default_value": 0.5,

There is no "Bluer" profile in the Cura source code.  "meshfix_maximum_resolution" exists as a variable name in there someplace, but the value for it is set from fdmprinter and altered by the Two Trees Base definition file overrides.  You just need to open that file in a text editor and change the "0.25" to "0.40".  You could also eliminate that override and it would default back to 0.5.  Another option is to add an override to your Bluer definition file.  That's where I'd put it and then save a copy off to the side for when Cura updates again.