Ultimaker/Cura #12342: Unable to Open Files

giebnatious, 2022-05-24:
### Application Version

5.0.0

### Platform

linux

### Printer

Creality Ender 3

### Reproduction steps

Open Cura 
click folder icon or File > Open File(s)
Nothing happens

### Actual results

Absolutely nothing

### Expected results

File explorer opens to select file

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

file:///media/fuse/crostini_d67ed83a6390db03b3bb253a67d29ab649d606cf_termina_penguin/.local/share/cura/5.0/cura.log

Goodfeat, 2022-05-26:
[image]
On my windows, most of the inputs do not want, writes an error

chipmunk1977, 2022-05-29:
This appears to be the same issue that I raised with the Beta version.  https://github.com/Ultimaker/Cura/issues/11874
There is a workaround - drag and drop works to open a file, and the gcode can be saved to a removable drive.  Not ideal, as the save doesn't allow renaming and automatically overwrites a previous file of that name, but apart from that the use of the system is unimpaired.

giebnatious, 2022-05-29:
I was able to open it by dragging and dropping however I was unable to save it at all. I went back to 4.13.1.

akshimassar, 2022-06-06:
This seems to be the duplicate of https://github.com/Ultimaker/Cura/issues/12266

Goodfeat, 2022-06-07:
> Похоже, это та же проблема, которую я поднимал в бета-версии. #11874 Существует обходной путь: перетаскивание работает, чтобы открыть файл, а gcode можно сохранить на съемном диске. Не идеально, так как сохранение не позволяет переименовывать и автоматически перезаписывает предыдущий файл с таким именем, но в остальном использование системы не ухудшается.

I don’t want to open this file at all, neither by dragging it, nor by using the file-> open tab

In general, I noticed that if the file is saved in binary format, then it opens, if left in ACSII, then the cura 5 refuses to open it