Ultimaker/Cura #12343: Cura 5.0.0 AppImage unable to start

PavelStoyan0v, 2022-05-24:
### Application Version

5.0.0

### Platform

Zorin OS 16 (Ubuntu)

### Printer

Ender 3 V2

### Reproduction steps

I am trying to run the AppImage for 5.0.0 but Cura doesn't start.
I see that the files in .local/share/cura are separated for each version so probably it isn't a conflict between the old and the new versions. In any case the previous version that I am still using is 4.13.0. 

### Actual results

The loading screen shows and after Loading UI it just disappears.
If I execute the appimage in terminal the final line says:
**QQmlComponent: Component is not ready**

### Expected results

Expected cura to start.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8765834/cura.log)

karcaw, 2022-05-25:
same issue here

v6-cblais, 2022-05-28:
Seeing the same issue as well on Zorin16. Found that executing with sudo permitted it to run.

nallath, 2022-06-14:
We've not been able to figure out what is going wrong here, so we will need more information. 

Could someone provide us with the lines of logging above "QQmlComponent: Component is not ready" and all the lines there when cura does run when with Sudo?

C-Blais, 2022-06-14:
Happy to help!

Without sudo:
$ ./Ultimaker-Cura-5.0.0-linux.AppImage 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
QApplication: invalid style override 'adwaita' passed, ignoring it.
	Available styles: Windows, Fusion
QQmlApplicationEngine failed to load component
QQmlComponent: Component is not ready

With sudo:
# ./Ultimaker-Cura-5.0.0-linux.AppImage 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QObject::startTimer: Timers cannot be started from another thread
Cyclic dependency detected between "file:///tmp/.mount_Ultimatua5mC/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_Ultimatua5mC/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext

JavaWantaBe, 2022-06-16:
I've got the same issue. What can I provide that would help troubleshoot?

eloo, 2022-06-17:
same here with Manjaro with Gnome on Wayland :(

nallath, 2022-06-17:
It might be somewhat related to this issue: https://github.com/Ultimaker/Cura/issues/11856

eloo, 2022-06-17:
> It might be somewhat related to this issue: https://github.com/Ultimaker/Cura/issues/11856

Not sure but imho the mentioned issue is tackling a different problem which comes upfront of this issue. 

I had the issue with OpenGL first and passed it with the mentioned workarounds. 
But know I'm stuck at this issue here

Ziiga, 2022-06-21:
I was experiencing the same issue but was able to start Cura with the following workaround: `unset QT_STYLE_OVERRIDE ./Ultimaker-Cura-5.0.0-linux.AppImage`.

System information:
OS: Manjaro Linux x86_64
Kernel: 5.10.123-1-MANJARO
DE: GNOME 42.2

eloo, 2022-06-21:
@Ziiga Thanks!

That actually worked!

PavelStoyan0v, 2022-06-21:
@Ziiga Thanks!
Works on Zorin OS 16.

peteruithoven, 2022-07-20:
If I understand Github correctly @casperlamboo's fix is included in the latest 5.1 release. But I still noticed I needed to manually run `unset QT_STYLE_OVERRIDE`. 

I'm running elementary OS 6.1 Jólnir (Built on Ubuntu 20.04.3 LTS). 

Is there a way for me to check whether the `unset` in the AppRun script was called? 

Logs:
```
./Ultimaker-Cura-5.1.0-linux.AppImage 

(AppImageLauncher:1022002): GLib-CRITICAL **: 15:04:24.412: g_file_test: assertion 'filename != NULL' failed
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
QApplication: invalid style override 'adwaita' passed, ignoring it.
	Available styles: Windows, Fusion
QQmlApplicationEngine failed to load component
QQmlComponent: Component is not ready
```

Casper, cool to see you're working for Ultimaker these days :grin:

eloo, 2022-07-24:
Can confirm that with Ultimaker-Cura-5.1.0-linux-modern we still need to unset qt_style_override

Canaill51, 2022-07-26:
Same on ZorinOS 16.1, need to unset qt_style_override.
Looking at here https://github.com/Ultimaker/Cura/commit/19d62da737dead7e2957703872bd5ecc038ac29e, seems merging has been rejected... 
Can someone confirm?

ovehbe, 2022-08-02:
The problem is `unset QT_STYLE_OVERRIDE` should be at the top or else it will not work for eos6 and probably some other distros.

This works for now;

[AppRun.txt](https://github.com/Ultimaker/Cura/files/9238871/AppRun.txt)

You can fix it like so;
`AppImage=./Ultimaker-Cura-5.1.0-linux.AppImage`
`./${AppImage} --appimage-extract`
Change squashfs-root/AppRun's code with code of the file above.
`./appimagetool-x86_64.AppImage squashfs-root` (note you might need to download AppImageTools)
`mv -f Ultimaker_Cura-x86_64.AppImage ${AppImage}`

jeremiah-k, 2022-08-20:
Reporting that unset QT_STYLE_OVERRIDE is not working for me using Ultimaker-Cura-5.1.0-linux-modern.AppImage in Linux Mint 21 and the error is the same with our without it.

```
unset QT_STYLE_OVERRIDE
./Ultimaker-Cura-5.1.0-linux-modern.AppImage
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimadYgBGK/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimadYgBGK/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
```

JayRugMan, 2022-08-25:
*Ultimaker-Cura-5.1.0-linux-modern.AppImage* did not work for me unless I run with ```sudo``` or with ```unset QT_STYLE_OVERRIDE```. Running Mint 21 as well. 

```bash
 ./Ultimaker-Cura-5.1.0-linux-modern.AppImage
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
QApplication: invalid style override 'kvantum' passed, ignoring it.
	Available styles: Windows, Fusion
QQmlApplicationEngine failed to load component
QQmlComponent: Component is not ready
^C
```
```bash
 unset QT_STYLE_OVERRIDE; ./Ultimaker-Cura-5.1.0-linux-modern.AppImage 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaNADGmW/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaNADGmW/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
```

closed 2022-10-11 by jellespijker

jellespijker, 2022-10-11:
Seems to be fixed by @Tim453 and should be in our next 5.2 release, I might post a link to a nightly to a 5.2.0-beta.2 later one for you guys to test

jellespijker, 2022-10-12:
If you could try the AppImages from this build and report if it actually fixes the issue, you would make me a happy developer.
https://github.com/Ultimaker/Cura/actions/runs/3232797852

peteruithoven, 2022-10-12:
Ultimaker-Cura-5.2.0-beta.2+27-linux-modern worked for me on elementary OS 6.1 Jólnir (Built on Ubuntu 20.04.5 LTS). No more need for `unset QT_STYLE_OVERRIDE`. :tada: 
Thanks for all the work on this!

Ziiga, 2022-10-12:
Ultimaker-Cura-5.2.0-beta.2+27-linux-modern.AppImage now also works on Manjaro without `unset QT_STYLE_OVERRIDE`.

C-Blais, 2022-10-12:
Zorin 16.1

Standard linux looks to be good. Seeing the following with the modern image:

~/Downloads/Ultimaker-Cura-5.2.0-beta.2+27-linux-modern$ ./Ultimaker-Cura-5.2.0-beta.2+27-linux-modern.AppImage 
[188638] Error loading Python lib '/tmp/.mount_Ultimas41Nyz/libpython3.10.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /tmp/.mount_Ultimas41Nyz/libpython3.10.so.1.0)

jellespijker, 2022-10-12:
@C-Blais that is a known issue you encounter with the modern image. Which is compiled on a system with a newer glibc version (compared to your system). Unfortunately there was an ABI breakage of that system lib, which forced us to create two different AppImages for newer/rolling distro's Ubuntu, Debian, Arch, Manjaro and older distro's

Which is not ideal but it is the most pragmatic solution at this moment for our Linux users.

C-Blais, 2022-10-12:
@jellespijker Thanks for the clarification! It would appear it is working as expected for me then.