Ultimaker/Cura #12346: Brim is printing on interior of model despite Only Outside checked.

mooklepticon, 2022-05-25:
### Application Version

5.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Sliced model with Brim only on outside checked.

### Actual results

Brim is showing on interior parts
[image]

### Expected results

There should be brim on the outer circle but not the inner gear stars.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

project file:
[gear_bearing sliced.zip](https://github.com/Ultimaker/Cura/files/8767362/gear_bearing.sliced.zip)

GregValiant, 2022-05-25:
What you have there is 8 models and each has the brim on the outside only.  Notice that the center piece does not have a brim on the inside of the hex hole.
That will print fine with a skirt.
I used mesh tools to split the model into it's component meshes.  "Brim on Outside Only" can't be set per model.  With the models split up it looks like this (but this can't be assembled.)  So is this a bug?  Not my call.  Someone from the Cura team will take a look.  I have seen this before on combined models though.
[image]

mooklepticon, 2022-05-25:
@GregValiant 
Interestingly, your image doesn't have brim on the inside of the outer gear, but mine does. Ignore the star gears.
[image]

Additionally, I thought this might be a 5.0 vs 4.X issue, but 4.13 does it, too.

Either I'm mis-remembering (possible) or the last time I printed this years ago, it only had them on the very outside.

GregValiant, 2022-05-26:
The brim on the inside of the outer gear didn't disappear until I moved all the little gears out of the center.  This might be considered a bug by the Cura Team.  I don't know if it's fixable for that STL file though.  Cura is seeing 8 models and treating them like 8 models.

mak0t0san, 2022-05-30:
I'm also seeing this behavior, even for just one model.
[image]

GregValiant, 2022-05-30:
@mak0t0san: in Cura use the File | Save Project command and zip and post the 3mf file.  The Cura Team will need it to recreate the problem.

nallath, 2022-05-31:
It's a known issue. The areas that do get the brim also have something that gets a brim on the outside. As you can see, the letters with a piece inside it do get a brim.

Ghostkeeper, 2022-06-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/4222.

closed 2022-06-01 by Ghostkeeper