Ultimaker/Cura #12348: Standby Temperature is wrong

garrinfelber, 2022-05-25:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Custom (Tenlog TL-D3 Pro)

### Reproduction steps

Select standby temperature of 149C for extruder 2

### Actual results

149C is only set in two places in the G-code:
1. Beginning of print when setting T0 to active temp
2. After second extruder is done and will no longer be used.

The G-code that is created (with my comments) is:
G1 X142.48 Y150.116 E47.29557 ;Final position for T1
G1 F4500 E12.29557 ;35mm extraction for tool-change
G92 E0 ;Reset T1 position to 0
T0 ;select tool 0
G92 E0 ;Reset T0 position to 0
M104 T1 S203 ; *** Incorrect standby temperature *** This is what's set as the "initial printing temperature"
M105 
M109 S183 ;T0 "initial temp"
G0 F7200 X154.686 Y138.249 Z1.2 ;Layer change
G0 X146.685 Y137.706 ;Initial print position
M104 S181 ;Set temp to "printing temperature"
;TYPE:SUPPORT
G1 F4500 E20.41575 ;Prime extruder
...

### Expected results

Expected Standby temperature to be 149:
T0
G92 E0
M104 T1 S149

Please fix. Thanks!

[bugreport_cura5_standbytemp.zip](https://github.com/Ultimaker/Cura/files/8769465/bugreport_cura5_standbytemp.zip)
[bugreport_log_standbytemp_cura500.zip](https://github.com/Ultimaker/Cura/files/8769499/bugreport_log_standbytemp_cura500.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Hello

GregValiant, 2022-05-25:
They aren't normally accessible.  Go to the Marketplace and load the Printer Settings plugin.  After re-starting Cura the Printer Settings group should be visible down at the bottom of the settings list.
To the right of the Settings Search box is an icon with three lines on it.  It's the Settings Visibility button.  Set the visibility to "All".

garrinfelber, 2022-05-25:
Cool, thanks. I just found the .json file with the setting in it.

I tried a larger model FYI and the temperature standby setting was done
correctly. I'll close the bug report, thanks so much for your help!

closed 2022-05-25 by garrinfelber