Ultimaker/Cura #12350: Cura 5 Error Slicing Model

rczook, 2022-05-25:
### Application Version

5.0.0

### Platform

Windows 10 Pro

### Printer

Ender 3 Pro

### Reproduction steps

1.  Loaded model
2.  Changed Z scale to 500%
3.  Moved model down 13mm in the Z direction
4. Clicked "Slice"

### Actual results

Get an error stating "Slicing failed:  Slicing failed with an unexpected error.  Please consider reporting a bug on our issue tracker."  Model works fine in Cura 4.13.1.

[image]

### Expected results

Model should slice and let me print.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This model slices fine in Cura 4.13.1.  Cura 5.0.0 seems to take quite a bit longer to open than Cura 4.13.1 does and takes longer to slice as well

[cura.log](https://github.com/Ultimaker/Cura/files/8771573/cura.log)
[Uploading SouthDakota_Z500_Z-13mm.zip…]()
.

GregValiant, 2022-05-27:
Can you re-try zipping and posting a Project File?

rczook, 2022-05-27:
[SouthDakota_Z500_Z-13mm.zip](https://github.com/Ultimaker/Cura/files/8786881/SouthDakota_Z500_Z-13mm.zip)

GregValiant, 2022-05-27:
It looks like there is something wrong with that "Dynamic" quality setting.

If you change to "Standard" quality and make the layer height .16 (like the Dynamic quality) then it slices.

Ghostkeeper, 2022-06-01:
The model slices correctly for me in the Cura 5.0 release as well as in the current main branch.

[image]

closed 2022-06-10 by nallath