Ultimaker/Cura #12351: Error En Cura 5.0 MODO ESPIRALIZADO

Marcelobarzo, 2022-05-25:
### Application Version

5.0.0 Beta+1

### Platform

Windows 10

### Printer

Core XY

### Reproduction steps

Intento imprimir una caja de espesor  0.5mm en formato espiralado (solo contornos). Con la versión anterior de Cura 4.13.1 funcionaba perfectamente y podía hacer unacajita con base de una capa de 0.5 mm en una pasada y las paredes las hacía de forma espiralada (contínua sin deteerse ni cambiar de dirección) y yo obtenía una cajita de aprox de 23 mm x 36 mm de base y 35 mm de altura con una sola pasada con pared de 0.5 mm de espesor en tan solo 8 minutos.

### Actual results

En esta version nueva de Cura Imprime la base siempre en espiral (aunque yo defina la base en forma de líneas), luego comienza a hacer las paredes en forma espiralizada (todos los contornos siguiendo un único sentido sin detenerse para cambiar de sentido) hasta la capa 4 o 5 en donde al imprimir una capa entera de pared cambia el sentido y comienza a imprimir nuevamente. 

En 

### Expected results

En la vista previa muestra un proceso completamente diferente mostrando un cruce en diagonal en cada altura de capa
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

GregValiant, 2022-05-25:
Load that model, make your settings, go to the File menu and use the Save Project command.  Zip the 3mf file and post it here.

Marcelobarzo, 2022-05-25:
[Envases.zip](https://github.com/Ultimaker/Cura/files/8774693/Envases.zip)

Thank you for the prompt response. Best regards.

GregValiant, 2022-05-26:
It is not a proper model for Spiralize.  It needs to be solid through-and-through like a calibration cube.  Because it is an open box it won't do a proper spiral because it has outside walls facing outward and outside walls facing inward.
What you can do is to turn on "Remove all Holes".  That will make it a solid model.  Then Spiralize will work.

Marcelobarzo, 2022-05-28:
It works perfectly. Thank you very much and sorry for my mistake.

closed 2022-05-30 by nallath