Ultimaker/Cura #12353: Wacom tablet problems - continued

3dHawwy, 2022-05-25:
### Application Version

5.0

### Platform

Mac Monterey 12.4

### Printer

Creality Ender 3 V2

### Reproduction steps

I am have problems using the Wacom tablet with Cura 5.0.  This is a new issue, never had problems with earlier Cura versions.  Sometimes it works for a minute or so, then Cura doesn't respond anymore to any Wacom control.  

### Actual results

I have to re-start Cura and using trackpad control or mouse ( which I don't like ).

### Expected results

Full Wacom support

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

Ghostkeeper, 2022-06-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/2515.

closed 2022-06-01 by Ghostkeeper

MariMakes, 2022-08-23:
Whoops.
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12554 instead.