Ultimaker/Cura #12355: Seam corner preference 'none' not working....keeps corner priority

PhilBaz, 2022-05-26:
### Application Version

5.0

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

Seam corner preference 'none'  and set coordinates for center of object.

### Actual results

Seam lands on corner

### Expected results

Seam should land on object wall....near the center of the wall in this case.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[Seam corner preference 'none'.zip](https://github.com/Ultimaker/Cura/files/8776003/Seam.corner.preference.none.zip)

Asterchades, 2022-05-26:
The mainstream build of Cura will only ever put seams in corners. The preference is simply to dictate which _type_ of corner. Yes, that does make the presence of both "Hide or Expose Seam" and "None" superfluous given they both achieve the same result, and I suspect that's a legacy hold-over from a time where things may have been different.

What does look unexpected in that picture, however, is that the seams on the front piece aren't all on the _same_ corner the way they are on the back. Given the settings you've shown they should all be on the inner edge. Cura has been struggling with that for longer than just 5.0, however - especially on "rounded" surfaces - for reasons I can only guess.

smartavionics, 2022-05-26:
FWIW, my cura builds are capable of placing the z-seam not on a corner when the seam preference is set to user defined and the corner preference set to none.

smartavionics, 2022-05-26:
[image]

smartavionics, 2022-05-26:
Here's an example of how this feature can help, compare these two images...

[image]

[image]

PhilBaz, 2022-05-26:
[image]

Asterchades, 2022-05-26:
I would imagine this report should be sufficient - it's the same people that deal with both, after all. It's not unreasonable to expect that two different options will yield two appreciably different results, even if sometimes similar, and "None" to me does feel like it shouldn't be preferring corners at all (though not actively avoiding them).

As I already mentioned, though, there's something funky with Cura's seam location decisions in general. Two identical layers shouldn't produce seams in different places unless the seam is explicitly configured as such (ie Random). Obviously this design doesn't have identical layers but something that fundamental wouldn't be helping in this situation either.

PhilBaz, 2022-06-25:
Has this been picked up by the Cura team ?

Seam placement in general is one of the biggest issues with Cura 5.

[image]

HOLLY-DION, 2022-06-25:
I hope a normal seam position is a straight or diagonal line, but this left-right-right arrangement actually makes me a little overwhelme
[image]
d

Joeydelarago, 2022-06-29:
Hello, we will discuss this seam issue. I agree the seam should be placed the same as earlier versions with this setting configuration.

Ghostkeeper, 2022-07-13:
Developers, see issue CURA-9474.

tlhintoq, 2022-07-22:
[image]

Personal guess... Seam placement insists on disabling the option to even choose "none".  Its insists on trying to place at "back".
Which I suspect it looks for where the *back* is in real time while slicing and comes up with a different number on every layer, so its all over the place.

In 4.13 we could actually set "none".  In 5.0/5.1 we can't.
[image]

jellespijker, 2022-11-16:
see also the FR in CuraEngine https://github.com/Ultimaker/CuraEngine/issues/1702

MariMakes, 2024-05-30:
Quick update, we are looking into ways to improve this for the upcoming release.
You can already take a sneak-peak here https://github.com/Ultimaker/CuraEngine/pull/2087