Ultimaker/Cura #12359: Could not probe OpenGL - Windows 10, Cura 5.0

Manchine, 2022-05-26:
### Application Version

5.0

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Installed Cura 5.0
Run as standard user - Error: "Could not Probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video drivers"

### Actual results

Installed Cura 5.0
Run as standard user - Error: "Could not Probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video drivers"
Run as Admin user - Error: "Could not Probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video drivers"

Check drivers in Device Manager - confirmed to be up to date (reinstalled just to be sure) - same error

Installed OpenGL Extension Viewer (6.3.7.0) - OpenGL version confirmed to be 3.1

Run "sfc /scannow" from CMD prompt (as Admin) + restart. Same error

### Expected results

Cura 5.0 should run normally

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Note: Cura 4.13.1 runs fine

just-trey, 2022-05-27:
Same issues on Ubuntu 22.04 LTS
AMD® Radeon rx 460 graphics
OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.0.1

planetar, 2022-05-30:
### Application Version
5.0 AppImage

### Platform
Arch Linux

### Steps to Repro:
download appImage, 
make it executable
run it

### Actual Result:
a msg box appears, "Failed to probe OpenGL" / Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.

starting cura from the cli I see
```

PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_cura_5cfBqJL/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/crocus_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_cura_5cfBqJL/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/crocus_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_cura_5cfBqJL/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast

```
Checking my OpenGL version:
```
glxinfo | grep OpenGL
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) HD Graphics 4600 (HSW GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.0.4
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 22.0.4
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.0.4
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

```

Cura 4.13.1 has no issue so I simply stay with the previous version as a workaround.

alexban011, 2022-06-01:
Also happens on arch 5.18.1

jukkaaakko, 2022-06-03:
> ### Application Version
> 5.0 AppImage
> 
> ### Platform
> Arch Linux with KDE
> 
> ### Steps to Repro:
> download appImage, make it executable run it
> 
> ### Actual Result:
> a msg box appears, "Failed to probe OpenGL" / Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.
> 
> starting cura from the cli I see
> 
> ```
> 
> PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
> libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_cura_5cfBqJL/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/crocus_dri.so) (search paths /usr/lib/dri, suffix _dri)
> libGL error: failed to load driver: crocus
> libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_cura_5cfBqJL/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/crocus_dri.so) (search paths /usr/lib/dri, suffix _dri)
> libGL error: failed to load driver: crocus
> libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_cura_5cfBqJL/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
> libGL error: failed to load driver: swrast
> ```
> 
> Checking my OpenGL version:
> 
> ```
> glxinfo | grep OpenGL
> OpenGL vendor string: Intel
> OpenGL renderer string: Mesa Intel(R) HD Graphics 4600 (HSW GT2)
> OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.0.4
> OpenGL core profile shading language version string: 4.60
> OpenGL core profile context flags: (none)
> OpenGL core profile profile mask: core profile
> OpenGL core profile extensions:
> OpenGL version string: 3.1 Mesa 22.0.4
> OpenGL shading language version string: 1.40
> OpenGL context flags: (none)
> OpenGL extensions:
> OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.0.4
> OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
> OpenGL ES profile extensions:
> ```
> 
> Cura 4.13.1 has no issue so I simply stay with the previous version as a workaround.
There is a work round to run Cura 5.0: Extract the appimage and renamel ibstdc++.so:

`./Ultimaker-Cura-5.0.0-linux.AppImage --appimage-extract 
cd squashfs-root/
mv libstdc++.so.6 libstdc++.so.6_appimage
./AppRun `

./AppRun should start Cura 5.0 wihtout issues.

nirnex, 2022-06-13:
I'm experiencing the same issue under Garuda (arch variant).
I'm getting the following error whenever I attempt to run Cura 5.0:Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.Here is the OpenGL information being returned by glxinfo from command line:
`OpenGL vendor string: AMDOpenGL renderer string: AMD Radeon RX 6700 XT (navy_flounder, LLVM 13.0.1, DRM 3.46, 5.18.2-zen1-1-zen)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.1.1
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.1.1
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.1.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20`

My system is fully up do date at this time, and other OpenGL based applications don't seem to be having issues. At this time Cura doesn't seem to be getting far enough to create a log file, however if I run it from commandline I see the following output from stderr/stdout:

`PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_UltimaweDIKM/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsilib
GL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_UltimaweDIKM/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsilibGL 
error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltimaweDIKM/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast`

shaneottinger, 2022-06-30:
Same error PopOS, I am noticing v 4.6 of opengl. Did they change how you look at versions with that? older versions seem to work fine.

shane@pop-os:~$ glxinfo | grep OpenGL
OpenGL vendor string: AMD
OpenGL renderer string: AMD RENOIR (LLVM 13.0.1, DRM 3.44, 5.17.15-76051715-generic)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.0.1
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.0.1
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.0.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

`No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy`

jellespijker, 2022-07-15:
Looks like a duplicate of #11856  which is fixed for 5.1 stable.

If any of you can confirm that the problem is fixed with the AppImage for Linux-latest that would help us out. See the build artifact in this run: https://github.com/Ultimaker/Cura/suites/7374973998/artifacts/299758926

hegjon, 2022-07-18:
> See the build artifact in this run: https://github.com/Ultimaker/Cura/suites/7374973998/artifacts/299758926

I am getting this error `Failed to generate URL to download artifact.` when I follow the URL

Zincr0, 2022-07-19:
Tested today with ubuntu 22.04 with the latest stable Cura 5.1 appimage from the web, same error:

```
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: no se puede abrir el archivo del objeto compartido: No existe el archivo o el directorio (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: no se puede abrir el archivo del objeto compartido: No existe el archivo o el directorio (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: no se puede abrir el archivo del objeto compartido: No existe el archivo o el directorio (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```

ChaosBlades, 2022-07-19:
I just tried 
Ultimaker-Cura-5.1.0-linux-modern.AppImage
on
Pop!_OS 22.04 LTS x86_64

Don't have any issues. I was able to slice something.

Zincr0, 2022-07-19:
Tested Ultimaker-Cura-5.1.0-linux-modern.AppImage, I can also confirm that at least it loads.

Josephur, 2022-07-29:
I also have this issue on a Windows 10 system that runs an Intel i5-2500s CPU, which is an older CPU (Intel HD Graphics 2000), however the previous versions of Cura worked fine.  Also the CPU does support OpenGL 3.0, and there are DLL entry points for backwards compatibility for Open GL 2.0 in the driver, so it should work, however it does not.

Something has to be wrong with the way it checks for OpenGL.. or this error message isn't providing enough detail and something else is funky.

jcforbes57, 2022-08-10:
I too am having this issue. Windows 10.

OpenGL 2.1.0 is installed:
`GL_VENDOR: Intel
GL_RENDERER: Intel(R) HD Graphics
GL_VERSION: 2.1.0 - Build 8.15.10.2900
GL_MAX_TEXTURE_SIZE: 8192
GL_MAX_COLOR_ATTACHMENTS: 8
GL_MAX_TEXTURE_IMAGE_UNITS: 16
GL_MAX_TESS_GEN_LEVEL: 0
`

batch111, 2022-10-26:
> Same issues on Ubuntu 22.04 LTS AMD® Radeon rx 460 graphics OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.0.1 Cura 4.13.1 has no issue

Thanks Old realease 4.13.1 works for me without error

elephantwalker50, 2022-12-22:
this continues to be a problem with cura 5.2.1, but works for cura  4.12.1,  windows 10 pro , latest build 19044, also see realtec extension viewer report for opengl (3.1 100% is running):

OS Name	Microsoft Windows 10 Pro
Version	10.0.19044 Build 19044
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	ENG-LF02
System Manufacturer	Dell Inc.
System Model	Vostro 2420
System Type	x64-based PC
System SKU	To be filled by O.E.M.
Processor	Intel(R) Core(TM) i3-2328M CPU @ 2.20GHz, 2200 Mhz, 2 Core(s), 4 Logical Processor(s)
BIOS Version/Date	Dell Inc. A12, 2/24/2014
SMBIOS Version	2.7
Embedded Controller Version	1.02
BIOS Mode	Legacy
BaseBoard Manufacturer	Dell Inc.
BaseBoard Product	0HTR08
BaseBoard Version	A12
Platform Role	Mobile

open gl extensions viewer report:

Renderer: Intel(R) HD Graphics 3000
Vendor: Intel
Version: 3.1.0 - Build 9.17.10.4459
Shading language version: 1.40 - Intel Build 9.17.10.4459

Max texture size: 8192 x 8192
Max vertex texture image units: 16
Max texture image units: 16
Max geometry texture units: 0
Max anisotropic filtering value: 16
Max viewport size: 8192 x 8192
Max Clip Distances: 6
Max samples: 4

GL Extensions: 129

GL_ARB_half_float_pixel
GL_EXT_compiled_vertex_array
GL_WIN_swap_hint
GL_EXT_packed_depth_stencil
GL_EXT_texture_lod_bias
GL_INTEL_map_texture
GL_ARB_color_buffer_float
GL_ARB_texture_compression_rgtc
GL_ARB_draw_instanced
GL_EXT_draw_buffers2
GL_EXT_texture_shared_exponent
GL_ARB_draw_elements_base_vertex
GL_ARB_window_pos
GL_ARB_explicit_attrib_location
GL_ARB_texture_float
GL_ARB_point_parameters
GL_ARB_texture_compression
GL_EXT_transform_feedback
GL_ARB_depth_buffer_float
GL_ARB_vertex_buffer_object
GL_EXT_packed_float
GL_ARB_pixel_buffer_object
GL_ARB_fragment_program_shadow
WGL_EXT_swap_control
GL_ARB_shading_language_100
WGL_ARB_extensions_string
GL_EXT_packed_pixels
GL_EXT_blend_func_separate
GL_INTEL_performance_queries
GL_EXT_texture3D
GL_ARB_texture_env_crossbar
GL_EXT_stencil_wrap
GL_ARB_transpose_matrix
GL_EXT_texture_compression_s3tc
GL_ARB_vertex_program
GL_EXT_texture_sRGB
WGL_EXT_pixel_format_packed_float
WGL_EXT_swap_control_tear
GL_ARB_texture_rectangle
GL_ARB_depth_clamp
GL_ARB_framebuffer_sRGB
GL_ARB_texture_env_add
GL_ARB_texture_rg
WGL_EXT_extensions_string
GL_ARB_half_float_vertex
GL_SGIS_texture_edge_clamp
GL_ARB_texture_cube_map
GL_EXT_shadow_funcs
GL_SGIS_texture_lod
GL_EXT_texture_filter_anisotropic
GL_EXT_multi_draw_arrays
GL_EXT_texture_rectangle
GL_ARB_vertex_array_object
GL_ARB_shadow
GL_ARB_vertex_array_bgra
GL_ARB_compatibility
GL_SGIS_generate_mipmap
GL_EXT_blend_minmax
WGL_ARB_pixel_format
GL_ARB_point_sprite
GL_ARB_depth_texture
GL_ARB_copy_buffer
WGL_ARB_make_current_read
WGL_ARB_pixel_format_float
GL_3DFX_texture_compression_FXT1
WGL_ARB_pbuffer
GL_ARB_map_buffer_range
GL_ARB_occlusion_query
WGL_ARB_create_context
GL_ARB_fragment_shader
GL_ARB_draw_buffers
GL_ARB_texture_border_clamp
GL_EXT_clip_volume_hint
GL_ARB_vertex_shader
GL_EXT_fog_coord
GL_ARB_fragment_program
GL_EXT_framebuffer_blit
GL_ATI_separate_stencil
GL_ARB_texture_env_combine
GL_EXT_abgr
GL_EXT_bgra
GL_ARB_texture_non_power_of_two
GL_EXT_texture_edge_clamp
GL_ARB_sync
GL_ARB_texture_query_lod
GL_EXT_separate_specular_color
GL_EXT_framebuffer_object
GL_EXT_texture_integer
GL_NV_conditional_render
GL_ARB_draw_buffers_blend
GL_EXT_blend_color
WGL_EXT_depth_float
GL_EXT_gpu_program_parameters
GL_ARB_timer_query
GL_NV_texgen_reflection
GL_EXT_blend_subtract
WGL_ARB_multisample
GL_EXT_secondary_color
GL_EXT_texture_swizzle
GL_ARB_provoking_vertex
GL_NV_blend_square
GL_EXT_texture_env_combine
GL_EXT_draw_range_elements
GL_EXT_texture_env_add
GL_ARB_multisample
GL_ARB_shader_objects
GL_ARB_texture_env_dot3
GL_NV_primitive_restart
GL_EXT_rescale_normal
GL_EXT_blend_equation_separate
GL_ARB_vertex_type_2_10_10_10_rev
WGL_ARB_buffer_region
GL_ARB_texture_buffer_object_rgb32
GL_ARB_instanced_arrays
GL_IBM_texture_mirrored_repeat
GL_ARB_uniform_buffer_object
GL_EXT_framebuffer_multisample
GL_EXT_stencil_two_side
GL_ARB_sampler_objects
GL_ARB_occlusion_query2
GL_ARB_framebuffer_object
WGL_ARB_framebuffer_sRGB
GL_ARB_multitexture
GL_EXT_texture_array
GL_ARB_seamless_cube_map
GL_EXT_texture_snorm
GL_ARB_texture_rgb10_a2ui
GL_ARB_fragment_coord_conventions
GL_ARB_shader_bit_encoding
Core features
v3.0 (100 % - 23/23)v3.1 (100 % - 7/7)v3.2 (70 % - 7/10)v3.3 (70 % - 7/10)v4.0 (21 % - 3/14)v4.1 (0 % - 0/7)v4.2 (0 % - 0/13)v4.3 (0 % - 0/20)v4.4 (0 % - 0/10)v4.5 (0 % - 0/11)v4.6 (0 % - 0/11)vARB 2015 (0 % - 0/12)OpenGL driver version check (Current: 0, Latest known: ):
Latest version of display drivers found
According the database, you are running the latest display drivers for your video card.

pietjongkoen, 2023-01-09:
This continues to be a problem with cura 5.2.1, but works for cura 4.12.1, windows 10
[image]

Grey-Lancaster, 2023-01-12:
Same issue here Cura 5.2.1

[image]

Grey-Lancaster, 2023-01-12:
I just went through all this and it worked
https://gist.github.com/rb-dahlb/26f316c5b6089807a139fc44ee69f0d1

[image]

mennucc, 2023-03-08:
I got this problem in Ubuntu 22.04 with _UltiMaker-Cura-5.3.0-linux.AppImage_ ; but _UltiMaker-Cura-5.3.0-beta.1-linux-modern.AppImage_ works fine.

mer-at, 2023-04-22:
Got the same Problem with UltiMaker-Cura-5.3.0-linux.AppImage on Debian 12.
UltiMaker-Cura-5.2.2-linux-modern.AppImage worked without Problems.

$ glxinfo | grep OpenGL
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon RX 6600 (navi23, LLVM 15.0.6, DRM 3.49, 6.2.10-x64v1-xanmod1)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.3.6
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.3.6
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.3.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

bullestock, 2023-05-18:
> There is a work around to run Cura 5.0: Extract the appimage and rename libstdc++.so:

For me, on Endeavouros (Arch) with Cura 5.3.1, this fix makes the error message go away - but the splash screen hangs forever at "Loading UI". The "modern" AppImage has the same problem when run without modifications.

fmgrotepass, 2023-10-11:
This problem occurs with the 5.5-beta1 for me. The 5.4 modern works without problems

Ortined, 2024-02-02:
> Я только что прошел через все это, и это сработало https://gist.github.com/rb-dahlb/26f316c5b6089807a139fc44ee69f0d1
> 
Спасибо большое, мне этот способ тоже помог!

Zogar89, 2024-02-22:
Same problem here with Intel 3000 Integrated and CURA 5.6

This works with my problem:
https://gist.github.com/rb-dahlb/26f316c5b6089807a139fc44ee69f0d1

snydeja, 2024-03-04:
> https://gist.github.com/rb-dahlb/26f316c5b6089807a139fc44ee69f0d1
> 
> to install the fix named curafix.sdb I copied to c:\ and ran from an admin cmd prompt sdbinst.exe -q c:\curafix.sdb

This worked for me too!