Ultimaker/Cura #12360: Cura installation for the first time give an error and wants i reset all my profile  loosing them. Do You Agree?

tmirante, 2022-05-26:
### Application Version

5.0

### Platform

windows 10

### Printer

anycubic mega 

### Reproduction steps

nothing i reistalled it but the error is the same. any other installation of cura goes ok

### Actual results

nothing error remain

### Expected results

to not loose all my profiles and setting of printers and filament and ..............peace in Europe

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

Asterchades, 2022-05-26:
The log you didn't include (despite hitting the check box) would help greatly in determining why it's doing this.

tmirante, 2022-05-30:
this is . errors available
[cura.log](https://github.com/Ultimaker/Cura/files/8797344/cura.log)

Ghostkeeper, 2022-05-30:
This is a problem with the approach that the Settings Guide used up until 2 years ago to display articles for things that aren't settings (such as the Troubleshooting pages). It's not a bug in Cura, but in a third-party plug-in. So I'll have to close this issue.

To resolve this issue for yourself, I suggest the following things:
* In the list of printers, delete the one named settings_guide_stack. If it's not there, I suggest that you go to Help -> Show Configuration Folder and then close Cura. Then delete the `settings_guide_stack.global.cfg` file in the `machine_instances` folder. Then start Cura to see if it's resolved.
* Update the Settings Guide to version 2.9.0. The version you're using (2.2.1) doesn't support Cura 5.0 yet.

closed 2022-05-30 by Ghostkeeper

tmirante, 2022-06-01:
ok closing but solution not works . can you say me how to update Setting guide ?