Ultimaker/Cura #12361: cura failing to slice large files with unexpected error

NiceManiac, 2022-05-26:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ender 5 Plus

### Reproduction steps

1. added large file
2. orient file correctly to fit
3. press slice

### Actual results

unexpected error occour after almost finishing slicing

### Expected results

file should be sliced and ready to print

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8780330/cura.log)
[helmet-1-piece.zip](https://github.com/Ultimaker/Cura/files/8780333/helmet-1-piece.zip)

GregValiant, 2022-05-26:
Turning off Monotonic and turning off Ironing allowed it to slice.  That print really doesn't need either as the only skin is that 15 x 15 patch on the top.
Somehow there is a conflict in there from those settings.  The Cura Team will take a look.

pearson222, 2022-05-27:
I am having the same issue - our files aren't the issue here...neither is the monotonic/ironing setting condition...

nallath, 2022-06-10:
I've been able to reproduce the issue. I will discuss it with the rest of the team!