Ultimaker/Cura #12362: Cura 5 Not Slicing Complex Models

centralcamsol, 2022-05-26:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality CR-10S

### Reproduction steps

Downloaded 3D model from thingiverse
Opened fresh install of Cura 5.0.0
Loaded file into Cura
Successfully sliced model
Changed Top/Bottom/Wall line count settings

### Actual results

Slicing process was slow, and eventually failed - threw error: Slicing failed with an unexpected error. 

Changing back settings, turning off arc welder, changing various other settings does not fix issue. 

### Expected results

Model should have been sliced for 3D printing.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

 - I was able to find this in the log file: 2022-05-26 10:46:04,089 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [954]: Backend exited abnormally with return code 3221225477!
 - Restarting Cura, restarting Windows has not solved the issue
 - Issue not repeatable on smaller/less complex model
[__Error.zip](https://github.com/Ultimaker/Cura/files/8780385/__Error.zip)
s

nallath, 2022-06-10:
I will discuss this with the rest of the team. Thanks for reporting this!

SharkWipf, 2022-07-21:
Can confirm this bug on version 5.1.0-beta+304. Some models need very specific wall thickness to be able to slice (i.e. 0.8 might error but 0.88 might work).  
The thickness that works seems to vary with line width.  
The model in question where I've experienced this has a lot of oval/circular 4mm thick walls. It also seems to interact with the infill setting, and the type of infill. In my case gyroid triggers it, cubic does not.

So, in conclusion: Most likely an issue with (gyroid?) infill in thin objects.

rburema, 2022-08-19:
This was added to a list of bugs to check if they had a similar cause (it seems they are all different though). In this particular case: However, it seems that error.zip does not contain any models, so we can't reproduce this unfortunately. -- We'd need a project file, and stl, or a set of reproduction steps to make a failing model to reproduce this, otherwise we won't be able to diagnose.

SharkWipf, 2022-08-19:
I'm not near any cura-loaded PC right now, or I'd provide one (not the OP though), but I've run into the issue a few more times and narrowed it down to more specific conditions to reproduce it:

- Gyroid infill
- Very thin infill sections
- Rounded shapes
- The error seems to depend on the infill offset, moving the object and reslicing can cause the issue to (dis)appear.

So without further error logs or models, I'm guessing the issue happens when gyroid infill is generated in a tight space at a specific curve and/or angle.
If OP or anyone else can't provide a STL/3MF file I'll try to generate one, but I won't be able to for at least a week and a half.