Ultimaker/Cura #12364: Error of segmentation.

Carlitoos95, 2022-05-26:
### Application Version

4.13.1

### Platform

windows 10

### Printer

creality ender 3

### Reproduction steps

hello 
i want to make some pieces and i import some stl o similar formats and all okey but when i click the segmentation button 
[image]
.

### Actual results

after that i go see preview to check everything is all right and there is some missing parts of the piece
[image]
 

### Expected results

i wanted to finalise the process because with this issue i cant make my pieces , if you wanna know i use SketchUp and i export it in STL formats but i tried in other formats and i can´t reach what i want. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

if it is my own error because im new please write me an email to kranoox@gmail.com showing me the option to fix this

GregValiant, 2022-05-26:
It looks to be a problem with the model, or that flange is too thin to slice.
You can try using MS 3D Builder to repair the model.  SketchUp often has reversed surfaces and open seams that won't allow parts to slice or that slice in error.

Carlitoos95, 2022-05-28:
i tried with that my guy 
but no result i´ll try with other CADs...

closed 2022-06-10 by nallath