Ultimaker/Cura #12367: 4.13.1 Profile With Inerior Threads Does Not Print Well When Sliced With 5.0.0

reynol3232, 2022-05-27:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Artillery X1

### Reproduction steps

I've made several hundred prints from a profile originally sliced in 4.13.1. The print has large internal ACME threads.  When I slice the same print in Cura 5.0.0 using the profile from from 4.13.1 the threads show a lot of stringing.  Photo #1 is the printed model sliced in Cura 5.0.0 and Photo #2 is the same printed model using 4.13.1.  All of the setting were identical between 5.0.0 and 4.13.1.  

### Actual results

Threads will fail to print when sliced in 5.0.0

### Expected results

A profile from 4.13.1 should print when sliced in 5.0.0.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Attached photos of prints that were sliced with identical settings in 4.13.1 and 5.0.0.  Cura 5.0.0 is having trouble printing overhangs
[image]
[image]
.

GregValiant, 2022-05-27:
There are new settings in 5.0 that had no equivalents in 4.13.  They need to be adjusted.
One that may apply here is "Wall Order".  The default is "Outside to Inside" which on overhangs isn't good.  Try changing it to "Inside to Outside" and do a test.

reynol3232, 2022-05-28:
You are correct!  The default setting in Cura 5.0.0 is opposite of what the setting is in Cura 4.13.1.  That is really bad.  When installing 5.0.0 it should have looked for the setting "Outer Before Inner Walls" from 4.13.1 and copied that setting to the new named "Wall Order" in 5.0.0.  This will indeed frustrate a few people.   I will try another print and see if that corrects the issue.

closed 2022-06-10 by nallath