Ultimaker/Cura #12369: support wall prints at double speed

exec79, 2022-05-27:
### Application Version

5 beta

### Platform

Windows 10

### Printer

voron 2.2

### Reproduction steps

setup support to 140mms.

### Actual results

it prints support at 280mms.  (feedrate at 100%, checked)

### Expected results

support prints at 140mms

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[CFFFP_Fach_mittelarmlehne_Focus3.zip](https://github.com/Ultimaker/Cura/files/8786071/CFFFP_Fach_mittelarmlehne_Focus3.zip)

GregValiant, 2022-05-27:
In the Speed section set the "Flow Equalization Ratio" to 0%.
It looks like it's being affected by "Gradual Support Infill Steps".  With the Flow Equalization at 100% and Gradual Steps at 3 then yes, it wants to print the support infill at 2X speed.  With either setting at "0" it doesn't happen.

I don't know what the intent was behind the Flow Equalization setting, but there may be issues with the implementation.

exec79, 2022-05-27:
> In the Speed section set the "Flow Equalization Ratio" to 0%. It looks like it's being affected by "Gradual Support Infill Steps". With the Flow Equalization at 100% and Gradual Steps at 3 then yes, [...]

thank you, thats it.

closed 2022-05-27 by exec79