Ultimaker/Cura #12370: cura 5.0 Marketplace extentions not working

Brombeerkralle, 2022-05-27:
### Application Version

5.0.0

### Platform

Windoes 10

### Printer

ASX1

### Reproduction steps

1. Upgrade to cura 5.0.0
2. uninstall cura 4.9.?
3. delete all cura 4.x.x files that were still in the cura files (explorer)
4. Uninstall all plugins
5. reinstall all plugins
6. none of theme were installed/ show up in the extention tab

### Actual results

They did not appear in the extention tab as in cura 4.x.x but marketplace marks them as installed

### Expected results

They shoult appear in the extetntion tab

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fieldOfView, 2022-05-27:
Could you try to restart Cura after uninstalling the plugins, then reinstall them and restart Cura again?

Brombeerkralle, 2022-05-27:
> I'm not Shure if I've tried uninstalling the plugin but I'll try.

Hmm, Yes, that helped. After uninstalling all, restarting Cura, reinstalling them and restarting cora once more they finally appeared.

The interesting thing there is that, after installing 5.0.0 I had to uninstall and reinstall all of them, so I'd still consider it a bug, but as one that can be fixed by the user if he knows how.

fieldOfView, 2022-05-27:
It seems to be necessary to restart Cura between uninstalling and reinstalling a plugin. If you uninstall, reinstall and then restart Cura, Cura can get in a state where it thinks a plugin is installed (as shown in the Marketplace), but the files are not actually in the right place in the configuration folder.

fieldOfView, 2022-05-27:
> I'd still consider it a bug

I consider it a bug too. I just had not gotten around to investigate if this also happens with other plugins than my own.

Brombeerkralle, 2022-05-27:
> I consider it a bug too. I just had not gotten around to investigate if this also happens with other plugins than my own.

It does.
I also installed Calubration Shapes by 5axes and had the same problem there.

Oh and btw, thanks for all those plugins! They really help!

Sonicboom247, 2022-05-27:
what i am having an issue with is the plugins that have been updated like 5axes auto orientation and a few others cannot be found via the marketplace, i also cannot locate the Cura\contents\plugins directory to manually add a plugin from its repository

fieldOfView, 2022-05-27:
> plugins that have been updated like 5axes auto orientation and a few others cannot be found via the marketplace

Some plugins simply have not yet been updated and/or approved for the Marketplace. The auto orientation plugin (by @nallath) is one of them.

>  i also cannot locate the Cura\contents\plugins directory to manually add a plugin

That has always been the wrong directory. If you want to install a plugin manually, go to the `plugins` folder inside the configuration folder; see Help -> Show configuration folder.

Sonicboom247, 2022-05-27:
> Some plugins simply have not yet been updated and/or approved for the Marketplace. The auto orientation plugin (by @nallath) is one of them.
>
>
> That has always been the wrong directory. If you want [...]

Thanks!! thats exactly what i was looking for, looks like the AutoOrientation was updated recently, i was able to copy it to the location and it is now working
[image]
.