Ultimaker/Cura #12372: Memory Leak, left on without processing for several hours allocated 75gb memory.

muckypaws, 2022-05-27:
### Application Version

5.0.0

### Platform

macOS 10.15.7

### Printer

Geeetech A20M

### Reproduction steps

Sliced a simple file, left Cura on Monitor and left for several hours.  Returned to my Mac and poor resonsiveness, looked at iStats (Screenshot attached), shows memory over 75gb.

[image]
 

### Actual results

Memory Leak clearly evident via iStats.  Quit Cura, and restart and memory allocation back to normal limits.

### Expected results

I wouldn't expect a memory leak
[cura.log.zip](https://github.com/Ultimaker/Cura/files/8789491/cura.log.zip)
[GA_Barrel_DualColour.3mf.zip](https://github.com/Ultimaker/Cura/files/8789496/GA_Barrel_DualColour.3mf.zip)
.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[GA_Barrel_DualColour.3mf.zip](https://github.com/Ultimaker/Cura/files/8789497/GA_Barrel_DualColour.3mf.zip)

fieldOfView, 2022-05-28:
Can you test if the same memory leak happens if you disable or uninstall the OctoPrint Connection plugin?

muckypaws, 2022-05-28:
Good Point, no it doesn't.  Does this point to an issue with OctoPrint Plug-in, or the interface to the plugin?

nallath, 2022-05-30:
> Good Point, no it doesn't. Does this point to an issue with OctoPrint Plug-in, or the interface to the plugin?

It's probably our age old nemisis zeroconf. The library that we use there seems to have a mem leak somewhere.

fieldOfView, 2022-05-30:
@muckypaws, can you open an issue at https://github.com/fieldofview/cura-octoprintplugin/issues? I'll see if I can get a solution for you there.

muckypaws, 2022-05-30:
Thank you! Will open it now, cheers :)

closed 2022-05-30 by muckypaws