Ultimaker/Cura #12373: Cura 5.0 Linux Appimage Crash after clicking slice

jk101290, 2022-05-27:
### Application Version

5.0.0

### Platform

Linux Mint 20.3

### Printer

Ender 3 Pro

### Reproduction steps

I was slicing a model after changing print speed to get time estimates for different speeds. After about 3 or 4 successful slices, the software closed out upon clicking "slice" the next time. Only parameter being changed was print speed before slicing. 

### Actual results

Cura closes.

### Expected results

Cura slices.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8789934/cura.log)

nallath, 2022-06-10:
Crashes how? 

I can see from the logs that it was unable to connect to the back-end at a certain point, but I don't see anything with regards to what could cause the crash.

Do you happen to have any VPN software (or simmilar) running?

jk101290, 2022-06-10:
Jaime,

The Cura application just closed with no error message. Yes I am running
VPN.

jk101290, 2022-06-13:
> Could you see if the issue happens if the VPN is disabled?
> 
> If the issue doesn't happen anymore, could you give more information about the VPN setup that you have?

I haven't been able to get it to crash again, with or without my VPN enabled.

closed 2022-06-14 by nallath