Ultimaker/Cura #12377: Print setting information window scroll is inaccessible

Teddy-1024, 2022-05-28:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

The information windows that appear when I hover over the print setting labels in the print settings window sometimes contain too much text to fit and have scroll bars, but when moving the mouse over to the information window, it disappears. Using the mouse scroll-wheel scrolls the print settings window and the arrow keys have no effect.

### Actual results

The information window disappears.

### Expected results

It would be helpful if the text in the information window could be viewed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

nallath, 2022-06-10:
This has been fixed in the final release of Cura 5.0. You are still using the Beta.

closed 2022-06-10 by nallath

Teddy-1024, 2022-06-10:
Awesome. Thank you very much for your help!