Ultimaker/Cura #12378: Bad slicing

Hum77, 2022-05-28:
### Application Version

5.0.0

### Platform

Win10

### Printer

Kywoo

### Reproduction steps

After slicing some kind of vase, with SPIRALISE slicer does some artefacts in preview. In a print too (same as in preview).
[image]

### Actual results

After slicing some kind of vase, with SPIRALISE slicer does some artefacts in preview. In a print too (same as in preview).

[image]

### Expected results

Slicing without artefacts inside volume of "vase".

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[vaza.zip](https://github.com/Ultimaker/Cura/files/8791451/vaza.zip)

closed 2022-06-10 by nallath

Hum77, 2022-10-11:
> The model has to be solid all the way through for spiralize to work. It
> can't be an "empty" vase.
> You can work with that one if you set Remove all Holes in the Mesh Fixes
> section. That will make it solid and then spiralize will work.
> Remember to turn it off when you are done.

Hi, thank you for feedback.
Strange is, that in a previous version it was ok with same settings.
Many greets, Petr Krejci.

GregValiant, 2022-10-11:
I looked at the differences between the 4.x versions and 5.x.
"Spiralize" by it's nature is a continuous extrusion that just keeps going up the model.  There should be no interruptions and the nozzle should not have to move from one "island" to another.  When there is a definite inner wall and a definite outer wall (two islands) then the nozzle must stop on one and move to the other.  That will leave a seam every time it happens.  An oddity is that the nozzle does not stop extruding when moving from one island to another.  That leaves a wall between the inner and outer walls.  The movement back to the start point is a travel move and so there is no extrusion when moving back.
So I decided that the 4.x versions were doing it wrong and the 5.x versions have gotten it right.

That doesn't mean everyone is happy with the new way that spiralize is handled because as you found, it's different.