Ultimaker/Cura #12379: Cura 5.0 hangs/crashes on multi-monitor setup

usernametakenaaaa, 2022-05-28:
### Application Version

5.0

### Platform

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

Main monitor: Nvidia GPU
Second monitor: Intel onboard GPU

Crash #1
1) Start Cura 5
2) Move the window to right side (Intel)
3) restart Cura 5 and it will permanently crash, also after reinstalling (config file is persistent)

Crash #2
1) Start Cura 5 (it opens on Main window, left, Nvidia)
2) Click on Marketplace and Cura will permanently hang/crash

Bug #3
1) Start Cura 5
-> A splashscreen will open on 2nd Monitor (Intel) while Cura opens on Primary monitor

Some Dialogs like Settings open correctly, some dialogs like "changes to profile" or "marketplace" open on second monitor with an immediate application crash

### Actual results

see reproduction

### Expected results

1) No hard crashes
2) do not splashscreen on wrong monitor
3) Uninstallation should also ask to remove configuration

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No logfile entries on the crashes

usernametakenaaaa, 2022-05-28:
I forgot to add:
To avoid the hard crashes:
1) Windows + P
2) Set "Second Monitor only" (windows sees the Intel always as primary even if not set as main)
3) start cura -> it will work fine now. As long as you don't dare use second screen

usernametakenaaaa, 2022-05-28:
One more update:
I've changed Intel to Nvidia dual GPU which is more common (so it is a normal dual monitor setup)
I also exchanged Monitor 1 and 2 so Windows correctly maps "primary" with "main" and "1"
The crashes are still persistent which means Cura does not work at all as soon as 2 monitors are connected to a PC, you might want to give at least one more monitor to the QA employee for the future

closed 2022-06-01 by Ghostkeeper

user897943, 2022-07-26:
I just got my 1st ever 3D printer and downloaded Cura after seeing many creators use it. This software is not even alpha if it cannot even be used with multi screens. I mean serious it's 2022 and this cr*p doesn't even have dual screen support, something that a lot of people will have such as laptop + external or anybody who's been working from home after the pandemic may have had their own screen + an additional one provided from their workplace. Fix this pathetic issue, developers. I would die of shame for putting out something that can't even work with a basic setup like a dual monitor. Pathetic.

nallath, 2022-07-26:
> . Fix this pathetic issue, developers. I would die of shame for putting out something that can't even work with a basic setup like a dual monitor. Pathetic.

Goodbye!

cmp-nct, 2022-10-31:
This has not been completed, it crashes consistently.
Windows 10 as well as Windows 11.

ataweg, 2023-05-28:
see https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1200322553

PhotoJoeCA, 2024-02-24:
Late February 2024 checking in: Three monitors on a desktop (3080). . . this is still an issue as of today.