Ultimaker/Cura #12380: Brim, Skirt and Raft don't display in preview

EM-III, 2022-05-28:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

All Printers

### Reproduction steps

I sliced my model with with Brim, Skirt and Raft. None of the previews showed the Brim, Skirt or Raft. 

### Actual results

When I printed the part with Brim chosen, the brim did print.

### Expected results

The brim should show up in the preview.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Yes, I did check to make sure I was viewing the first layer.

nallath, 2022-05-30:
Did you also enable the "helpers" in the visibility screen of the preview?

[image]

EM-III, 2022-05-30:
Helpers is checked but still the supports are not displayed.

EM-III, 2022-05-30:
[image]

nallath, 2022-06-10:
I can't reproduce this on any of the models that I slice with. Could you provide a project file that reproduces the issue?

EM-III, 2022-06-10:
Without changing anything, When I now slice the exact same model file, the supports and brim/skirt now appear correctly. Before I submitted the original bug complaint, I did reboot, so I am at a loss why it now works and didn't before. 
[image]

closed 2022-06-10 by EM-III