Ultimaker/Cura #12382: Error file explorer Linux ChromeOS

bartstuurman, 2022-05-29:
### Application Version

5.0.0

### Platform

ChromeOS 102.0.5005.75 (Officiële build) (64-bits)

### Printer

Custom (Easythreed x1)

### Reproduction steps

Cura 5.0.0. starts but i cant open the file explorer to select an STL file, I can drag and drop an file in Cura and slice it. But then I am unable to save the file because the file explorer won't open.

### Actual results

File explorer wont open

### Expected results

File explorer should open to select or save a file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no log because nothing happens...

UndeadKernel, 2022-05-29:
I experience the same exact problem when using the cura AppImage in Arch GNU/Linux.

akshimassar, 2022-06-06:
This seems to be the duplicate of https://github.com/Ultimaker/Cura/issues/12266

closed 2022-06-07 by nallath