Ultimaker/Cura #12383: Nozzle move to Prime Tower position even when function is disabled

scharrrfi, 2022-05-29:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

XTLW Climber 8

### Reproduction steps

Disable prime tower in the menu.

### Actual results

When I change the extruder, the nozzle moves on every layer start to the prime tower position.

### Expected results

The nozzle should move directly to the print after parking.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://wetransfer.com/downloads/77d24e04d3029c47c886cc9d5961eb5620220529140917/58908fd90e2d13f248ddfcc5936c7fa820220529140931/df23b3

GregValiant, 2022-05-30:
I can't check that because your "XTLW Climber 8" is an unknown machine.  When I open the 3mf file I just get the model with no settings.
Check in the Experimental section and see if you have "Wipe Nozzle Between Layers" enabled.

scharrrfi, 2022-05-30:
Wipe Nozzle Between Layers is disabled. 
I attached you my imported printer configuration. 
[CURA 4.13.1 configuration file.zip](https://github.com/Ultimaker/Cura/files/8797604/CURA.4.13.1.configuration.file.zip)

GregValiant, 2022-05-30:
That was hard to find.  The problem is in the Extruder files so you need to go back to the manufacturer/author "XTLW 3D" and let them know.  There is also at least one typo in the definition file.  We'll let them find it.

Extruder 0 has these lines:
"machine_extruder_start_pos_x": { "value": "prime_tower_position_x" },
        "machine_extruder_start_pos_y": { "value": "prime_tower_position_y" },
        "machine_extruder_end_pos_abs": { "default_value": true },
        "machine_extruder_end_pos_x": { "value": -53 },
So whether you have the prime tower enabled or not the XY start position is at the location where the prime tower would be if it was enabled.  The movement is "absolute" and the X end position is "-53" and that's off the build plate.

The problem with Extruder 1 is similar and the extruder definition has these lines:
"machine_extruder_end_pos_x": { "value": 258 },
        "machine_extruder_end_pos_y": { "value": "prime_tower_position_y" }
So the same sort of thing is happening with the second extruder but it moves way to the right off the build plate.  That's why there are three points in the back of the build plate that the print head keeps wanting to go to.

After altering the extruder files to divorce them from the prime tower location and get rid of the start and end positions I got this which I think is how it should be when there is no prime tower:
[image]

scharrrfi, 2022-05-31:
Thank you! I will edit the files too and report to the manufacture.

closed 2022-05-31 by scharrrfi