Ultimaker/Cura #12384: Bug about add print and profile

uyminhtri2702, 2022-05-29:
### Application Version

5.0

### Platform

window 10

### Printer

mega Zero

### Reproduction steps

import profile and add printer new

### Actual results

erro
[image]

### Expected results

fix bug! help me

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-05-30:
Are you trying to install that with the stock Mega Zero definition in Cura, or did you get a custom definition file from someone?  It might be trying to load a couple of files that don't exist on your computer.
I don't have a problem installing a Mega Zero using the stock Cura definition (also Windows 10).

nallath, 2022-05-30:
Not only would I like an answer on the questions posted by greg, I'd also need your configuration folder and log files in order to debug the issue.

closed 2022-06-13 by github-actions[bot]