Ultimaker/Cura #12387: Tromxy X5SA 2E

andreinit, 2022-05-29:
### Application Version

5.0.0

### Platform

Windows 10 x64

### Printer

Tronxy X5SA 2E

### Reproduction steps

I sliced some 3d models and put them on the card. During printing, there was no heat applied to the nozzle head nor bed, even though in the program they were

### Actual results

Introduced sd, aligned Z, started printing but no set temperature was existing (had to manually set it)

### Expected results

It should have taken the temperature for the nozzle

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Unable to upload zip X_x

GregValiant, 2022-05-29:
Someone will need to look at that Gcode file to see what was going on.  Since it is a text file I think you could change the file name extension to ".txt" and post it.

andreinit, 2022-05-30:
[TXXPRO2ED01P_vive_Desk_Holder_no_end.txt](https://github.com/Ultimaker/Cura/files/8799459/TXXPRO2ED01P_vive_Desk_Holder_no_end.txt)

GregValiant, 2022-05-30:
I think it's the printer.  These lines are from your Gcode file and are exactly what should be in your startup gcode in the Cura Machine Settings.
**M190 S60** ; Set bed temperature and wait
G28 ; Home all axis
; Uncomment the line below to enable ABL Mesh probing
;G29 ; Probe bed mesh for ABL
; For best results do not run nozzle heater while performing ABL
G1 Z5.0 ; Raise nozzle to prevent scratching of heat bed
G1 X0 Y0 ; Move nozzle to Home before heating
**M109 S240.0 T0** ; Set nozzle temp and wait
So there it is:  bed at 60 and hot end at 240.  Pretty much all printers use these commands to set the temperatures.
You _are_ able to manually set the temperatures from the knob on the LCD?

andreinit, 2022-05-31:
Yes, I am able to set them manually and they do heat up

GregValiant, 2022-05-31:
Unzip the attached file and print the gcode file.  There is no model or extrusions - just heating commands.  It's as simple as I could make it.
It auto-homes and then heats the hot end and then the bed using M104 and M140 (regular heat commands)
It homes again and then heats the hot end and bed using M109 and M190 (the heat-and-wait commands).
There are a couple of beeps in there to let you know when a section has ended.
While it's "printing" watch your LCD and see if the temperatures change
[HEATING TEST.zip](https://github.com/Ultimaker/Cura/files/8807398/HEATING.TEST.zip)
.

andreinit, 2022-06-01:
Good day.
I watched the heating test through the LCD screen and the temperature changes, on the head to 150 C (kept 0 on bed) and then on the bed and 0 on the head. 😮

GregValiant, 2022-06-01:
The M104 and M140 commands are first, then the M109 and M190 "wait" commands heat the hot end and bed again.
So you should have seen:
auto-home
Hot end to 150
Hot end off
Bed to 50
Bed off
auto-home
Hot end to 150
Hot end off
Bed to 50
Bed off
auto-home

GregValiant, 2022-06-01:
Then that gcode should print.
I opened it in Notepad and changed just the temperatures so I could print PLA and then I printed the first layer with no problems.
I don't know what to tell you.  It prints on my Ender and should print on your TronXY.
Of course the settings weren't right for the Ender and centering is way off, but here is the first 2 1/2 layers.
[image]

closed 2022-06-10 by nallath

Altirix, 2022-07-25:
similar issue here, no clear reason as to what is happening. using the stock Tronxy XY-2 Pro profile on cura 5.0

The printer heats up bed to 60c, and just stays there with no error and it never continues, the hotend stays at room temp.

manually heating the hotend and bed works just fine, and hit the target temp. the gcode cura seems to create isnt doing what it should

[TXY2PROPRO2EXY3SE_Ender3 V2 Voron psu clip C.txt](https://github.com/Ultimaker/Cura/files/9182665/TXY2PROPRO2EXY3SE_Ender3.V2.Voron.psu.clip.C.txt)

GregValiant, 2022-07-25:
@Altirix  Within that startup gcode of the file you posted is:
M109 S250.0 T0 ; Set nozzle temp and wait
In Cura go to Manage Printers and Machine Settings and look in the StartUp Gcode box.  You will see that line as:
M109 S{material_print_temperature_layer_0} T0 ; Set nozzle temp and wait
You can try taking out the "T0" then click in the Ending Gcode window and then close the dialog.  Depending on how TronXY configured the firmware, maybe it doesn't like having the Tool Number called out.

Something else you could try would be to put an M104 line just before the M109 line:
M104 S{material_print_temperature_layer_0}
Maybe your printer is configured to require an M104 before it will respond to an M109.  It didn't require an M140 before the M190 though so this is another wild guess.

xaviergriffon, 2022-07-29:
Hello,
@andreinit , i have exactly the same problem with my X5SA PRO.
After several tests and comparisons of the gcode generated with Slic3r (which worked), the problem is in the commentary of the commands.
If you replace the S in lower case, the instructions work.
M190 S50.0 ; Set bed
M104 S200.0 T0 ; Set nozzle temp and wait
Replaced by :
M190 S50.0 ; set bed
M104 S200.0 T0 ; set nozzle temp and wait

I have no explanation but this works for me.

andreinit, 2022-07-29:
I've talked with the tronxy support team and they gave me a link to an update to the firmware. On initial testing, it worked well. Going to update after testing it a bit more, to make sure it was not a fluke