Ultimaker/Cura #12389: Mouse off-center since Cura 5 beta 1.

Hawkeye666, 2022-05-29:
### Application Version

5.0

### Platform

WIndows 11 and 10

### Printer

all

### Reproduction steps

When launching Cura 5 (since beta 1) the mouse cursor is usually slightly out of alignment with the cursor graphic, Usually the hot spot is lower and to the right by about 10 pixels each direction. 

### Actual results

move mouse cursor over a button or check box, 

### Expected results

Button should highlight, check box should toggle selected or not, but this does not happen. About once in every 5 to 10 times I run Cura 5 this works correctly. Been a problem since beta 1 and still a problem in released version. Never happened with any other version of Cura prior to 5.x.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none available
[cura.log](https://github.com/Ultimaker/Cura/files/8793989/cura.log)
.

nallath, 2022-06-10:
Would it possible for you to make a video recording of this behavior? We've not been able to reproduce this issue ourselves.

Hawkeye666, 2022-06-10:
I'll give that a shot and update. Something else I noticed this morning, this only happens when the app is windowed, if I run Cura 5 full-screen then the hot spot is aligned perfectly. Windows 1 22H1, triple monitor setup with latest nVidia drivers on RTX 2070. Same results on all monitors, even the one in portrait mode. I do have a couple of screen/video utilities running and I need to enable/disable those and test as well. Let me do that first and then I will post back here to let you know.

Hawkeye666, 2022-06-10:
that should read Windows 10 22H1.

Hawkeye666, 2022-06-10:
I found the problem! Though there was no conflict with Cura 4.x or 4.x there's definitely a GUI compatibility problem between Windowblinds 10.x from Stardock and Cura 5.x. Excluding the Cura EXE from being reskinned by WindowBlinds fixed the problem.

closed 2022-06-10 by Hawkeye666