Ultimaker/Cura #12390: Objects with holes are having the holes removed after slicin

gubbas8900, 2022-05-29:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 5 Plus

### Reproduction steps

Objects with holes are having the holes removed after slicing.  
1) Load object
2) Place face to bed (face with holes)
3) Slice
4) Holes gone

[image]
[image]
.

### Actual results

Preview shows no holes.  Printed object has holes deleted.

### Expected results

Object should appear as created.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE5P_Base.zip](https://github.com/Ultimaker/Cura/files/8794501/CE5P_Base.zip)

Asterchades, 2022-05-30:
If you don't want it to do it, don't tell it to do it. Uncheck "Remove All Holes" in Mesh Fixes and it will leave them in place.

gubbas8900, 2022-05-30:
What is interesting, is I never set that setting, nor have I ever needed to bother with the mesh fixes settings.  Now that you pointed it out, I unchecked and all is normal again.  Curious how it got checked.

Asterchades, 2022-05-30:
Have you downloaded a print or printer profile at some point? I'd be very curious to find out where and how this happened as well - it crops up far too regularly over on Reddit for it to be something that people have accidentally done.

gubbas8900, 2022-05-30:
Previously using Cura 4.13.1.  When I loaded 5 it copied over everything so I have not directly loaded anything into 5.

The only profile I had loaded into 4 was for my SeemeCNC Rostock Max.  Attached are the profiles given by them.
[SE300.zip](https://github.com/Ultimaker/Cura/files/8798623/SE300.zip)

nallath, 2022-06-10:
Since this was a case of misconfiguration, I'm going to close this.

closed 2022-06-10 by nallath