Ultimaker/Cura #12391: Top/Bottom pattern ignored when using spiralize outer contour

dalovering, 2022-05-30:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

1. Add part to be printed with vase mode (i.e. spiralize outer contour)
2. Set "Bottom layers" to 3
3. Set "Top/Bottom pattern" to "Lines" or "Zig zag"
4. Set "Spiralize outer contour" setting to enabled
5. Slice

### Actual results

Bottom layers are filled with "Concentric" solid fill pattern
[image]

### Expected results

Bottom layers should have shells according to "Wall line count" and "Lines" or "Zig Zag" solid fill, depending on "Top/Bottom pattern" setting, as in previous versions of Cura (4.12.1 shown)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[20mm cube project file.zip](https://github.com/Ultimaker/Cura/files/8794790/20mm.cube.project.file.zip)

gitlui, 2022-06-05:
I can second this behaviour. Here are two screenshots of the same model and same profile between cura 4.13 and 5.
In the screenshot left ist cura5 and right ist cura 4.13
[image]

PhilBaz, 2022-06-09:
See

https://github.com/Ultimaker/Cura/issues/12075

closed 2022-06-09 by nallath