Ultimaker/Cura #12395: Unable to find STL in Cura 5.0.0

TheFellmonkey, 2022-05-30:
### Application Version

5.0.0-beta+1

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

Went to file path my .STLs are stored in (which I can access without issue with Cura 4.13.0) but the file explorer in Cura 5.0.0 cannot see anything in the folder (either subfolders or .stls)

### Actual results

Unable to load stls into Cura 5.0.0

### Expected results

Folder structure to be navigable in the Cure 5.0.0 file explorer and Cura 5.0.0 to recognise stl files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

nallath, 2022-05-30:
Uh, that's weird. Is the extension of those files .STL? If so, does it work if you change it to .stl?

You reported the issue to be with 5.0.0-beta. Could you try if it does work with the stable 5.0.0 release.

TheFellmonkey, 2022-06-07:
Ah I hadn't spotted that the proper 5.0.0 release has come out since I downloaded the beta; that's fixed everything and it's all behaving normally again!

Many thanks nallath!

closed 2022-06-07 by nallath