Ultimaker/Cura #12396: Cura reboot bug

dps2k22, 2022-05-30:
### Application Version

5

### Platform

Mac OS monetary 

### Printer

Ender 3 copy 

### Reproduction steps

Every time I switch my Mac on i am having to reinstall cura 5, it works when installed 

i5 MacBook Pro running monetary

### Actual results

I have to install again after restart 

### Expected results

I should be able to enjoy cura every time I start my machine 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

N/A

nallath, 2022-05-30:
You didn't include the log files. We need them to diagnose the issue.

dps2k22, 2022-05-30:
Hi, sorry it doesn’t kick up any error logs I don’t think

dps2k22, 2022-05-30:
sorry, problem solved, was clicking the wrong icon