Ultimaker/Cura #12397: Adaptive layers doesnt account for support Z distance

ImagineerNL, 2022-05-30:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1- set default layer height: 0.14
2- set support Z distance 0.28 (exactly 2 layers)
3- set adaptive layers, keep default settings.
4- slice

### Actual results

Unless something else on the involved layers invokes a different layer height, the Z distance is now 0.36 instead of 0.28.
This is following the constraint that the Z distance is always a multitude of the current layer height (which is max 0.18)

[image]

### Expected results

The expected result is that the set default layerheight (of 0.14) should be treated as current layer height.

Another option:
add to the 'adaptive layers' a value 'layer height support clearance'

Even better would be:
- Support --> support Z Distance --> add field 'Z Distance # layers'
- 'Support Z distance' divided by 'Z distance layers' = layerheight for the layers clearing the support

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[adatpive_layers_support_clearance.zip](https://github.com/Ultimaker/Cura/files/8797974/adatpive_layers_support_clearance.zip)

GregValiant, 2022-06-11:
Using adaptive layers means different layer heights dependent on the geometry of the area being sliced.  Your .14 layer height is ignored as the geometry of the part indicated to Cura that there should be .18 layer heights for the two layers above the support.  So yes, the current layer height of .18 was used and the first skin above the support goes down .36 above the support.

The alternative to using .18 would be to abandon the Adaptive Layers during a transition over support so that the area-of-interest would be at .14 layer height.

I think we both agree that right now Cura is doing what we expected.  The "current layer height" is being used to calculate the gap over the support.  It would appear that you would like to see a different method of calculation for that so the gap is the distance of 2 layer heights as input in the settings rather than using the Adaptive Layer height above the support.

I'm inclined to change this to a Feature Request rather than a bug report.  Let me know what you think.

GregValiant, 2022-06-13:
A feature request it is.
I don't know what the developers will think of this, but a way to look at it could be that if there is Support Interface present on a layer then the Layer Height for those layers plus the next 2 layers would be at "Layer Height" and not at the "Adaptive Layer Height".
I would change the title to:  Adjust "Adaptive Layers" to use regular "Layer Height" over support gap/Top Distance.
@ImagineerNL  does that sound reasonable?

ImagineerNL, 2022-06-13:
@GregValiant then i would opt for a selfdefined layerheigt at the support interface settings and use that.
This would make it more 'readable' for end-users as otherwise you would still have the issue of a difference between the z clearance and the current layer height.

GregValiant, 2022-06-13:
Your comments will be in here.  I don't think you'll get yet a third layer height though.  Most users would find it entirely too confusing.   It would be one thing to make Support Interface layers = layer height.  It would be a whole 'nother ballgame to add some intermediate user defined layer height.
No matter what you do the roof over the support will still have a poor finish as there just isn't enough squish.

Ghostkeeper, 2022-07-08:
This issue has been known with us for a long time. I pointed it out when reviewing the Adaptive Layers feature when that got merged. We then created a ticket for it on our backlog and it's been on the backlog ever since. Developers, see CURA-5104.

The issue arises because CuraEngine tends to calculate vertical distance in terms of a number of layers. This goes for:

* Support Z Distance
* Support Overhang Angle (!)
* Support Interface Thickness
* Support Stair Stepping
* Gradual Infill
* Gradual Support
* Skin thickness
* Possibly a bunch of other features

We didn't write down an approach on how to fix it yet, but I don't like the third layer height setting either. I think a better solution would be for CuraEngine to properly calculate the correct number of layers to create the desired vertical distances. This wouldn't solve Skin Thickness since it has a setting for a fixed number of skin layers, but possibly that setting would have to be disabled if adaptive layers is enabled then.

ImagineerNL, 2022-07-08:
Calculating the layer thickness sounds like the best approach indeed.