Ultimaker/Cura #12400: Adaptive Layer height uses the wrong base layer height to start with

usernametakenaaaa, 2022-05-30:
### Application Version

5

### Platform

win 10

### Printer

Ulti S3

### Reproduction steps

The "Adaptive Layers" feature is using the first layer height for all it's calculations.
"Maximum Variation" for example is a setting of how much layer height might differ from Layer #1

The problem here is that Layer #1 is the Adhesion layer used for Brims, Rafts.
In my case I have a 0,05 layer height for the first layer and 0.2 layer height for the model

"Adaptive Layers" needs to base it's relative settings on the Model layer height "Layer Height" instead of "Initial Layer Height" which is only used for adhesion and typically deviates from the model layer height.

### Actual results

Changing "Initial Layer height" seems to be used as base for the layer heights of the "Adaptive Layers"

### Expected results

Changing "Initial Layer height" should not influence the layer heights of the "Adaptive Layers"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Replicate:
Use any model
Enable Adaptive Layers
Change Initial Layer height to 0.02 and to 0.3, you'll see how the entire model is changed based on that.

usernametakenaaaa, 2022-05-31:
To add to it: Adaptive layers is rarely working the way I would want. 
It's not that smart. It would be great if we could just put layer "zones" based on Z height as an alternative to it "manual mode" 
Often I know best which areas need a different layer height

nallath, 2022-06-10:
But it kind of makes sense that it first starts off with the initial layer height, right? Otherwise you could end up with an extremely thin initial layer, which is very likely to ruin your print.

As for the manual selection, you're right, but it would take quite a lot of work to implement. I believe we still have it on our backlog, but the time needed to implement it would simply be too much.

closed 2022-06-24 by github-actions[bot]