Ultimaker/Cura #12401: Master profile templates for custom profiles to reference

ericnd01, 2022-05-30:
### Is your feature request related to a problem?

If this is already possible, please let me know and I apologize.  I have several custom profiles created (fine print, draft print, supports, no supports, rafts, etc.).  Sometimes I need to change a setting (i.e. bed temp, print speed, etc.) that needs to be applied to all or some of the custom profiles.  Unless I'm missing something, I have to edit each profile to make these changes.  I know that bed temp can be set in the material profile, but using it as an example.

### Describe the solution you'd like

Would like to have something like "master profile templates" where the user chooses a setting that would be used in specific custom profiles.  The custom profiles would then be able reference the master profile setting if desired.  So, if I had custom profiles for fine, standard, and draft printing and the bed temp for all of these custom profiles references a master profile template.  If I were to add a glass bed, instead of changing the bed temp from 60 to 65 for each profile, I could just edit the master profile template bed temp setting and all profiles that reference it would automatically be updated.  Might also be nice to allow master profile templates to reference other master profile templates.  That way you could have a top level master template for things like bed temp, print speed, etc. and a sub template for things like print detail (which reference the master template), and a custom template that has things like supports, rafts, etc. which reference the sub template (which in turn references the master template).

### Describe alternatives you've considered

Tried Googling, but didn't find anything.

### Affected users and/or printers

Anyone using multiple custom profiles with common settings that need to change.

nallath, 2022-06-10:
There are a few plugins that you might want to check out. The universal settings plugin does something akin to this. You can also use the materials setting plugin to add more settings to the materials.

There is also the option to create your own profiles, where you obviously have a lot more power (but it is also easy to mess things up!). You can find more info about that here: https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings

5axes, 2023-03-09:
Nice Idea .. In my ToDo List
[image]