Ultimaker/Cura #12405: Weird slicer bug

Matszwe02, 2022-05-31:
### Application Version

5.0

### Platform

Windows 10

### Printer

Marlin 2.0.9.3

### Reproduction steps

Cura somehow combined model from previous print, destroying my print. Also printer halted without doing anything.
Lowered nozzle to ~3mm below print and stayed there until I stopped it.

### Actual results

[image]

### Expected results

[print.gcode.txt](https://github.com/Ultimaker/Cura/files/8809002/print.gcode.txt)

Cura log hasn't updated for 2 days, so posting it won't help

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Could you check what happened? And if my newly flashed marlin 2.0.9.3 makes any difference?

GregValiant, 2022-05-31:
I think it's a problem with the SD card(?)  The gcode file you posted ends abruptly with no ending gcode or "settings" area.
These are the last three lines in that gcode file:
G1 F3000 X187.715 Y101.702 E1551.22544
G0 F7200 X186.208 Y101.219
G1 F3000 X185.359

So the hot end and bed aren't being shut off, the steppers aren't disabling...all of that is missing.  that is usually indicative of corrupt sections on a memory card, but you didn't mention how you printed the file.

Matszwe02, 2022-06-01:
> I think it's a problem with the SD card(?) The gcode file you posted ends abruptly with no ending gcode or "settings" area. These are the last three lines in that gcode file: G1 F3000 X187.715 Y101.702 E1551.22544 G0 F7200 X186.208 Y101.219 G1 F3000 X185.359
> 
> So the hot end and bed aren't being shut off, the steppers aren't disabling...all of that is missing. that is usually indicative of corrupt sections on a memory card, but you didn't mention how you printed the file.
> 
> Within the gcode file this happens: 74607 G1 X106.897 Y90.209 E2677.9.34 Y71.857 74608 G1 F3000 X89.793 Y72.643 E1534.14211
> 
> The 74607 line would never have the Y value at the end like that. The E value in that second line is obviously not a continuation from the previous line and so must be from a different gcode file.

Thanks. Yes, I forgot to mention that I printed it from my SD card. Does corruption really causes the old gcode to stay?

GregValiant, 2022-06-01:
The corrupt sectors on the card interfere with the "Start of File" and End of File" markers on any file.  So your file didn't have an ending, and the other file didn't have a beginning so they just ran together.  The old file couldn't be deleted because it didn't have a beginning for the operating system to begin deleting it from.
Formatting the SD card to FAT32 usually fixes this problem.  I am still using the card that came with my Ender 3 Pro and I've had to format it a number of times.