Ultimaker/Cura #12406: Could not probe OpenGL, even though I have latest (5.15)

alexban011, 2022-05-31:
### Application Version

5.0.0

### Platform

Linux (arch 5.18.1)

### Printer

didn't make it that far

### Reproduction steps

I just opened the AppImage in the terminal. Not sure how you're supposed to reproduce this.

### Actual results

[image]

### Expected results

To open up normally

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have the latest proprietary drivers from Nvidia and the GPU is showing in lscpi -v. Other OpenGL applications run normally.

Additional info which was written in the terminal:
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
qt.qpa.wayland: EGL not available

I will also test on X11 to see if maybe its a wayland issue

alexban011, 2022-05-31:
Cura 4.13.1 from the store works. Its only a 5.0 issue

alexban011, 2022-05-31:
Also happens on X11

just-trey, 2022-06-01:
duplicate of #12359