Ultimaker/Cura #12408: Incomplete extrusions

j10mdza, 2022-06-01:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

Sliced this file in Cura 5.0.0 under standard profile settings for ender 3 v2.
[Uploading CE3_reaction_wheel_cube_2.zip…]()

### Actual results

There are incomplete extrusions in the top & bottom layers. It appears that the variable line width feature in the new engine is not working properly.
[image]
[image]

### Expected results

All extrusions should be complete and no gaps should be present throughout the layers.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_reaction_wheel_cube_2.zip](https://github.com/Ultimaker/Cura/files/8810215/CE3_reaction_wheel_cube_2.zip)

nallath, 2022-06-10:
I will discuss this with the rest of the team, thanks for reporting it.

As a bit more info; We currently only apply the variable line width on the walls (and not on the infill / top/bottom). That is why you're seeing these artifacts.

nallath, 2022-06-14:
I was partially correct about my statement! We do use the variable line width settings if you use concentric top/bottom filling. Using the variable line width with linear patterns would be quite computationally expensive and would be very slow to slice.

closed 2022-06-14 by nallath