Ultimaker/Cura #12419: Cura 4+5: Printer start GCODE is saved in project files and overwrites current printer start code when opened as project

Triangulix, 2022-06-01:
### Application Version

4.8.0 + 5.0.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Save a project
2. upgrade your printer and adapt the start GCODE
3. Load your saved project
4. Watch your print fail

### Actual results

Stupid behavior.

### Expected results

Since the printer configuration, including the printer start code, belongs to the printer and not to the project it is wrong to save it as such.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

People upgrade their printers and optimize their start code. It makes no sense at all to restore old start GCODE, written for a printer of the past, when opening an old project to print it again.

GregValiant, 2022-06-04:
I vote they leave it alone.  When you open a project file you have the option to "Update Printer" or to Create New Printer".  Selecting "Create New" insures that your personal settings like start and end gcode don't get replaced.
I altered my Start Up Gcode so the first line comment is ";Greg's Ender 3 Pro" and when I see that I know it's good to go.  If it's missing and says just Ender 3 then it was over-written when I opened a project file and forgot to select "Create New".  When that happens I open one of my own project files and my standard StartUp Gcode gets re-inserted.

Triangulix, 2022-06-04:
> I vote they leave it alone. When you open a project file you have the option to "Update Printer" or to Create New Printer". Selecting "Create New" insures that your personal settings like start and end gcode don't get replaced.

Why would you create "new printers" for every old printer or printer setup you don't own anymore?
I hope the Cura developers think harder about this issue before wiping it off the desk.

nallath, 2022-06-10:
The idea of a project is that it restores things to the state as close as possible. As such, it also includes the start / end gcode. 

If you simply want the models from a project file, you can always chose to only import those (and thus, not get any models loaded).

closed 2022-06-10 by nallath