Ultimaker/Cura #12422: start gcode edit not saving

TheCraftDude, 2022-06-01:
### Application Version

5,0 newest official release

### Platform

windows 10

### Printer

custrom printer skr 1,4

### Reproduction steps

restarts pc - opens cura - select printer- manage printers - machine  settings - (edits something in start gcode) - closes the machine settings windows when done) - settings not saved next time opening - does this 9 times in a row, no effect, has to edit file by hand.

### Actual results

cura acts like a child thats trowing a tantrum

### Expected results

cura to save start g code like it always did.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

only what i wrote is what i did

nallath, 2022-06-10:
I just tried this on master. Clicking the X without switching seems to work now.

Once 5.1 is out and you can still reproduce it, feel free to re-open or create a new issue!

MariMakes, 2022-08-22:
Hey @TheCraftDude,

Thanks for the report!

This seems to be a duplicate of
https://github.com/Ultimaker/Cura/issues/11941
We have an issue on our backlog with the intent to fix it. 

I'm closing this issue since it's a duplicate.

closed 2022-08-22 by MariMakes