Ultimaker/Cura #12423: Error slicing STL file

ffguzmanf, 2022-06-01:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Add model to cura
Slice model

### Actual results

Error 

### Expected results

Stl sliced
[project.zip](https://github.com/Ultimaker/Cura/files/8817926/project.zip)
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8817935/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8817939/cura.log)
[project.zip](https://github.com/Ultimaker/Cura/files/8817941/project.zip)
[image]

nallath, 2022-06-10:
I can reproduce the issue as well. I will discuss it with the team. Thanks for reporting it :)

GregValiant, 2022-07-05:
@nallath,  I opened this in 5.1 Beta and it still won't slice.