Ultimaker/Cura #12424: Buttons of the top right corner menu aren't working

Triffen, 2022-06-01:
### Application Version

5.0 appimage

### Platform

Linux Manjaro

### Printer

Ender 3

### Reproduction steps

Open Cura 5 appimage.
Click menu button in the top right corner.
Click any of the buttons inside.

### Actual results

Nothing happens.

### Expected results

Something should open I guess :) One of those buttons is "Report a bug" and it doesn't work either. Seems to be Linux and/or appimage bug.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2022-08-22:
Hey! 

Thanks for reporting. This seems to be a duplicate of
https://github.com/Ultimaker/Cura/issues/12232

This should be fixed in Ultimaker Cura 5.1.0 Linux modern.AppImage.asc
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.1.0

I'm closing this issue but please let me know if the issue is still present.

closed 2022-08-22 by MariMakes

Triffen, 2022-08-22:
It wasn't working for me in Cura 5.1 appimage. I have installed xdg-desktop-portal and now the buttons work, also save/open dialogues for files, projects and profiles also work.

MariMakes, 2022-08-22:
Huzzah! 🎉

bprzybytkowski, 2022-10-19:
The buttons still doesn't work for me using both 5.1 and 5.2 appimage on SteamOS (Arch Linux). This applies to `Import` button in Profiles window as well which hurts most as I am unable to import my profiles used on Windows machine

Triffen, 2022-10-19:
@bprzybytkowski Have you tried my solution above? It fixed this kind of problems to 5.1 and to 5.2 (5.2 won't slice for me though but that's a different issue).